reiniscimurs/DRL-robot-navigation #106: The area of environment

Ethan0207, 2024-01-18:
Hello,
It's me again. I have a problem about the project. What is the area of the training environment？
[image]

reiniscimurs, 2024-01-18:
You mean the size of it?
One square is 1x1 meters, so you could calculate based on that.

Ethan0207, 2024-01-18:
OK, thank you for your reply.

Ethan0207, 2024-01-18:
Hello,
I have another problem. In this project, what is the meaning of step, episode or epoch?

reiniscimurs, 2024-01-18:
See: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part3-training-13b2875c7b51

Ethan0207, 2024-01-19:
Hi，
Thank you for your reply. Maybe I hasn't understood the meaning of episode. In part3, it indicates that this is a collection of subsequent steps, until one of the termination conditions is reached. But when I train the agent, an episode is updated approximately every 20 minutes. And we can see the agent can reach the termination quickly. Could you explain it?

Ethan0207, 2024-01-19:
Hi，
Thank you very much. I know what you mean. Just like this sentence, is that means this is the 62th epoch, and the average reward is -82.113001? But what is the meaning of the "0.900000" and how do we know how many episodes it has gone through?
[image]

reiniscimurs, 2024-01-19:
That is the average reward during the 10 evaluation runs after epoch 62.
0.9 is the collision rate.
This would be quite easy to tell from the code: https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/train_velodyne_td3.py#L32-L37

I would suggest to fully familiarize yourself with the code. This will help understanding what is in this repo.

closed 2024-04-28 by reiniscimurs