reiniscimurs/DRL-robot-navigation #107: Change robot model

Ethan0207, 2024-01-22:
Hi,
These days, I would like to replace the robot model with my own four-wheel differential model. But I don't know how to replace it. Could you give me some suggestions? Where should I change?

reiniscimurs, 2024-01-22:
Hi,

You can launch your own robot from the launch file here: https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/assets/multi_robot_scenario.launch#L7-L11

Please see other issues on changing the robot model, like: https://github.com/reiniscimurs/DRL-robot-navigation/issues/82

Ethan0207, 2024-01-24:
Hi,
Thank you for your reply. I have changed my robot, but I meet some difficulties. When I use command " python3 train_velodyne_td3.py", in rviz, there is only the model. Other than that, nothing happened. And the terminal displays  errors.
[image]
[image]
This is my multi_robot_scenario.launch file.
[image]
This is my mini.gazebo.launch file.
[image]
This is my xacro file.
[image]
[image]
[image]
And in the velodyne_env.py, I just modified the vehicle name
[image]
[image]
So could you give me some suggestions?

reiniscimurs, 2024-01-25:
Hi,

in the first image you can see that you have an error. So it seems your subscriber is not getting any odom information from your subscriber. You should check your topic names and if they are subscribed to what they should be. Also, the issue might be in differential drive plugin so you should check that for your robot

Ethan0207, 2024-01-26:
Hi，
Thank you for your reply. Today,  I checked all the nodes and topics, and it seemed normal. But it don’t send "/cmd_vel" topics.  And I checked my differential drive plugin. It seems to be fine. I don't konw how to solve it. Could you give me some suggestions?
Thank you very much.
This is my differential drive plugin file.
[image]
[image]

Ethan0207, 2024-01-26:
Added:
When I first use command " python3 train_velodyne_td3.py", it prints many errors. And when I kill it, then use command " python3 train_velodyne_td3.py" again, it shows differently, but it still has errors.
[image]
[image]

reiniscimurs, 2024-01-27:
I would suggest to first try understanding the error messages yourself, backtracking them and debugging them. And only after you have sincerely tried to understand (and googled them) to ask about issues here. There are many google answers to your p gain errors (which are not really errors). You are asking questions about things that are not part of this repo and something that you should be working on your own end about your own changes. Do actually investigate on your own end before asking questions.

For the other issue, your odom value is None because your subscriber is not returning any values. As you can see from your topics in rostopic list, your plugin does not seem to use robot namespace. So your topic names do no match where you subscribe to them and where they are published.

Ethan0207, 2024-01-30:
Hi，
Thank you for your patient response. I had solved my problems. The problem was the namespace. However, from topic list and rqt, “goal_point”"linear_velocity""angular_velocity" didn't publish. From rviz, we can't see the target point and the movement path. What could be the reason for this? And in the code, where are the set target points and action tracks displayed?
[image]
[image]

reiniscimurs, 2024-01-30:
Are you sure they are not published? Because they are in your rostopic list. If you want to check if anything is published to a topic you can just read it at runtime with `rostopic echo your_topic_name`. If it outputs anything, it is eing published.

It seems to me that you are just not displaying anything in rviz. I suggest to familiarize yourself with rviz and follow some tutorials on how to display information there.

Ethan0207, 2024-01-31:
Thank you for your reply. I will try it.

closed 2024-04-28 by reiniscimurs