reiniscimurs/DRL-robot-navigation #109: Compatibility and Testing of ROS Melodic Branch

HazelFannn, 2024-01-31:
Hi,

I've been successfully running the main branch for ROS Noetic and now wish to use the Melodic branch. However, I've encountered issues establishing the simulation properly after several attempts. Could you please confirm:

1. Has the Melodic branch been thoroughly tested?
2. Are there any known issues or limitations with this branch?

Thanks for your help!

reiniscimurs, 2024-01-31:
Hi,

The melodic branch is deprecated and should be taken "as is" as I will not be able to test it nor update it. It should work out of the box without an issue if set up properly. In principle it is the same code as in Noetic version and was tested when implemented. However, if you encounter any issues I would probably not be able to help as I am not planning on supporting it and it is here for legacy reasons.

HazelFannn, 2024-01-31:
Thank you very much for the reply! I will try to adjust my setup again.

closed 2024-03-09 by HazelFannn