reiniscimurs/DRL-robot-navigation #112: Lose efficacy: removing all obstacles and navigating

hzxBuaa, 2024-02-09:
Hello, may I ask if it is normal for me to use a trained model in testing mode to find the endpoint, but after removing all obstacles and navigating on the open ground, I cannot find the endpoint?

reiniscimurs, 2024-02-18:
Hi,

Not sure what the setup here is. What do you mean with not being able to find the endpoint? Is it not able to navigate after removing all obstacles?

If that is the case, my guess is that all laser readings are returning maximum value (10 by default) which is a situation not encountered in training. As such, the policy does not know how to respond to such out of distribution scenarionand policy performance degrades.

hzxBuaa, 2024-03-07:
OK thank you very much for your answer

closed 2024-03-07 by reiniscimurs