reiniscimurs/DRL-robot-navigation #119: some questions about sim2real

nlnlnl1, 2024-03-04:
Hi. When I test the weights trained in the virtual environment on a real car. My real car has been spinning around since moving forward a distance, my real car can avoid obstacles but cannot reach the target point.
I can confirm that the input of the real car is aligned with the input in the virtual environment., But the real testing environment of the car is very different from the virtual environment. I'm wondering if the spatial structure of the real environment should be similar to that of the virtual environment.
Or do you have any other experience related to Sim2real?
Thank you!

nlnlnl1, 2024-03-05:
Okay, thank you for your answer. So, may I ask if you tested in a real environment on a map similar to a virtual environment?
Or rather, I haven't had enough time to train the model yet, but during my training process, all indicators have reached their peak and tested well in a virtual environment. I'm not sure if extending the training time will be effective in this situation.

reiniscimurs, 2024-03-05:
Hi,

You can see the environments here: https://www.youtube.com/watch?v=MhuhsSdzZFk&ab_channel=ReinisCimurs
They were very different than the training environment at test time.
Most likely there is some discrepancy between real sensors/robot/environment/transforms/etc. and the ones in the simulation. More training will probably not help. Make sure that the data is displayed the same way as in training and that goals and other values are given in the same range as in training.

closed 2024-04-28 by reiniscimurs