reiniscimurs/DRL-robot-navigation #123: cmake error

Leo6666666666666, 2024-03-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - ROS version:
 - Repository branch:

**Additional context**
Add any other context about the problem here.

reiniscimurs, 2024-03-10:
Hi, 
This is supporting software issue and outside the scope of this repo. You should find a way to install empy for your setup.

Sunsundayz1, 2024-05-14:
Hi

Sunsundayz1, 2024-05-14:
Hi
You should enter `catkin _ make -DPYTHON _ EXECUTABLE = / usr / bin / python3`
The conda environment conflicts with the python environment

cheeseburgercat, 2024-09-18:
Hi，i find this solution
`catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3`