reiniscimurs/DRL-robot-navigation #125: How to change the robot to different one?

am-amani, 2024-03-17:
Hello @reiniscimurs,

Thanks for your support so far. I have a question:

Could you please guide me how I can change the robot model? For example I want to use turtlebot 3 or any other mobile robot model. What kind of modification is required? Also I would like to use lidar, not velodyne. How workspace should be modified or replaced?

Thank you for your attention. Sincerely, looking forward to hearing from you.

reiniscimurs, 2024-03-18:
Hi,

Please see the tutorial and the turtlebot branch on how to use turtlebot robot model:
https://medium.com/@reinis_86651/using-turtlebot-in-deep-reinforcement-learning-4749946e1c15
https://github.com/reiniscimurs/DRL-robot-navigation/tree/Noetic-Turtlebot

Velodyne puck sensor is a lidar. If you want to change to a different sensor, you can add it to your robot (in p3dx case it would be here: https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/catkin_ws/src/multi_robot_scenario/xacro/p3dx/pioneer3dx.xacro#L27) and update the code according to the sensor you add. These changes will depend entirely on the sensor used.

am-amani, 2024-03-20:
@reiniscimurs, thanks for your reply.

I am a student, and there are other types of robots in our lab, such as turtlebot2i and ROSbot XL (4-wheel mobile robot). Therefore, I want to use these robots for the simulation training and test them on real robots. Can you guide me on the procedure for changing the robot? Should I try just like what you did here: https://medium.com/@reinis_86651/using-turtlebot-in-deep-reinforcement-learning-4749946e1c15
https://github.com/reiniscimurs/DRL-robot-navigation/tree/Noetic-Turtlebot

am-amani, 2024-03-20:
@reiniscimurs, I also noticed that one of my robots is available on ROS2. This is not going to cause any conflict, right?

am-amani, 2024-03-20:
@reiniscimurs, I also have another question. On turtlebo3 (all versions), I didn't see any camera available. I want to know how to add and use a camera for the model.

reiniscimurs, 2024-03-21:
Hi,

I can only try to answer specific questions about this repository. Any additional changes you want to make should be done and tested in your side. 

As far as robot models go, use the guide for how to call a different model. If you have a compiled robot model, it should be fairly straightforward. I do not know about ROS2 robots and if there are any changes to files themselves. They are spawned differently in ROS2 though.

Camera and any other sensor can be added as in the aforementioned comment.

am-amani, 2024-04-05:
Dear @reiniscimurs,

I successfully integrated a LiDAR sensor into both the Pioneer3x and TurtleBot3 AutoRace models and proceeded with their training. However, I'm encountering an issue when attempting to replicate this setup with the Waffel_pi model and another robot of similar dimensions. They exhibit unusual movement patterns as documented here:

Specifically, the robots' linear velocity fluctuates abruptly between 0 and 1, leading to unintended acceleration and a tendency to rotate along the x-axis. This behavior is not observed in the Pioneer3x and TurtleBot3 AutoRace models. Could this issue be related to the robots' height, or might there be another underlying cause? Any insights or suggestions you could offer would be greatly appreciated. I tried with different lidars, but no results.

Additionally, I've attached a video recording that captures the described movement in more detail:

https://github.com/reiniscimurs/DRL-robot-navigation/assets/33548412/805cd216-a273-4577-b6b6-646a0878f466
https://github.com/reiniscimurs/DRL-robot-navigation/assets/33548412/1b7aaf68-6c6d-4574-8861-ae5eb9ea8a41

Thank you for your assistance.

reiniscimurs, 2024-04-05:
Hi,

AutoRace is not a model type. Do you mean turlebot3 Burger model?

Judging by the videos your state constantly ends in a collision so a reset is made. Most likely your sensor reading is detecting a self collision. You should place the sensor where it will not detect a collision with the robot model or update the code to filter out such situations. Regarding the height, it is a factor if you are using a 3D velodyne LiDAR. For that, see the explanation here: https://medium.com/p/4749946e1c15

reiniscimurs, 2024-04-05:
Here you can see that your laser is detecting your own model base:
[image]

am-amani, 2024-04-22:
Thanks for your reply dear @reiniscimurs !

closed 2024-04-22 by am-amani