reiniscimurs/DRL-robot-navigation #126: Hello, thank you for your work. After training for about 110 times, I started Killall -9 ROSOUT ROSLANCH ROSMASTER gzserver nodelet robot_state-publisher gzclient python python3 and stopped training. However, when I used the python3 test_velodyne-td3. py command, I observed that the car only performed circular motion without path planning. May I ask why this phenomenon occurred and how to solve it? Thank you! Looking forward to your reply

hhhiiittt, 2024-03-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - ROS version:
 - Repository branch:

**Additional context**
Add any other context about the problem here.

hhhiiittt, 2024-03-19:
Hello, thank you for your work. I trained 110 times and then stopped the training using the terminal command you provided. There are also saved training models in the folder. But when I tested, the car could only perform circular motion without path planning. What is the reason for this? How to solve it? Thank you for your reply!
[image]
[image]

hhhiiittt, 2024-03-19:
[image]

reiniscimurs, 2024-03-19:
Hi,

See other issues on convergence:
https://github.com/reiniscimurs/DRL-robot-navigation/issues/103

hhhiiittt, 2024-03-19:
> Hi,
> 
> See other issues on convergence: #103

thank you

reiniscimurs, 2024-03-19:
Btw what do you mean by 110 times? Is it 110 episodes or epochs?

hhhiiittt, 2024-03-19:
> Btw what do you mean by 110 times? Is it 110 episodes or epochs?

Thank you for your reply.It is 110 episodes

reiniscimurs, 2024-03-19:
110 episodes is not enough training time for the model. From my experience you will need 40 to 60 epochs to start seeing any reasonable performance.