reiniscimurs/DRL-robot-navigation #128: [gazebo-1] process has died

hbonnavaud, 2024-03-28:
**Describe the bug**
Robot actions not visible in rviz and a message "[gazebo-1] process has died [pid ****, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode TD3.world __name:=gazebo __log:=/home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1.log]" is shown repetitively.

**To Reproduce**
I followed the steps here https://github.com/reiniscimurs/DRL-robot-navigation/tree/Noetic-Turtlebot?tab=readme-ov-file#installation

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 in a virtual box vm
 - ROS version: Noetic
 - Repository branch: Noetic-Turtlebot, bu had the same problem with main branch.

** screenshot **
[image]

**Additional context**
Add any other context about the problem here.

Here is the logs
```
started roslaunch server http://localhost:43235/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/publish_frequency: 30.0
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    urdf_spawner (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311/

process[gazebo-1]: started with pid [30678]
process[urdf_spawner-2]: started with pid [30682]
process[robot_state_publisher-3]: started with pid [30684]
process[joint_state_publisher-4]: started with pid [30685]
process[rviz-5]: started with pid [30686]
[INFO] [1711642598.487174, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1711642598.491574, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[gazebo-1] process has died [pid 30678, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode TD3.world __name:=gazebo __log:=/home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1.log].
log file: /home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1*.log
[gazebo-1] restarting process
process[gazebo-1]: started with pid [30821]
[gazebo-1] process has died [pid 30821, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode TD3.world __name:=gazebo __log:=/home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1.log].
log file: /home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1*.log
[gazebo-1] restarting process
process[gazebo-1]: started with pid [30874]
[gazebo-1] process has died [pid 30874, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode TD3.world __name:=gazebo __log:=/home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1.log].
log file: /home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1*.log
[gazebo-1] restarting process
process[gazebo-1]: started with pid [30915]
[gazebo-1] process has died [pid 30915, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode TD3.world __name:=gazebo __log:=/home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1.log].
log file: /home/robot123/.ros/log/8d24b920-ed1e-11ee-85f6-790f21794f25/gazebo-1*.log
```
and so on, the code is still running but rviz only shows a white robot in the center of a green room.

reiniscimurs, 2024-03-29:
Hi,

This seems to be a ROS specific problem and not directly related to the code in this repo. My best guess is that the VM is just not powerful enough to display rviz at runtime. I would suggest increasing the VM resources.

hbonnavaud, 2024-03-29:
The VM have access to intel core i7 + 16 Go RAM. Definitely not a resource issue.