reiniscimurs/DRL-robot-navigation #129: Unable to load the world to the Rviz

kikixiong, 2024-03-28:
**Describe the bug**
Fail to load the map to the RViz simulation

**To Reproduce**
I just try to run 
`cd ~<PATH_TO_FOLDER>/DRL-robot-navigation/TD3
python3 train_velodyne_td3.py`
follow the tutorial , but it seems there is some issues that it can not visulize and train as expected.

**Expected behavior**
The trainning simulation appear normally with the world with obstacles.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:Ubuntu'20.0.4
 - ROS version: noteic
 - Repository branch: *main
 - Python: 3.8.10
 - Torch: 1.10

**Additional context**
I have look into issues and comments on #16  but still i can not solve this problem.
By the way, I have successfully lunch the TD3.world in gazebo following the suggestion on #16

reiniscimurs, 2024-03-29:
Hi,

It seems like your resource path is not sourced properly. This will depend on where you have cloned the repo to. See the answers here: https://github.com/reiniscimurs/DRL-robot-navigation/issues/64

kikixiong, 2024-03-29:
I have follow the instruction on the #44 and tutorial to identify the source of gazebo and check other resourcec path.  It seems to have been set correctly by the command `export ROS_HOSTNAME=localhost
export ROS_MASTER_URI=http://localhost:11311
export ROS_PORT_SIM=11311
export GAZEBO_RESOURCE_PATH=~playground/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch
source ~/.bashrc
cd playground/DRL-robot-navigation/catkin_ws
source devel_isolated/setup.bash`
I also have use `echo $GAZEBO_RESOURCE_PATH` to check GAZEBO path before and after it seems to be correct as specify in the tutorial.
[image]

reiniscimurs, 2024-03-29:
Seems to me you are missing a `/`

Try: `GAZEBO_RESOURCE_PATH=~/playground/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch source`

kikixiong, 2024-03-29:
Thanks! With your advice, I've resolved the issue.  Since, I am not that familiar with ubuntu and gazebo ros so my question seem tobe a little bored. But thanks again for your timely response,your responce really help me a lot.

closed 2024-03-29 by reiniscimurs