reiniscimurs/DRL-robot-navigation #130: Gazebo environment（Problems encountered during training）

Heritagecll, 2024-04-07:
Hello! When I started training, it seemed like I was successful, but only rviz appeared without gazebo. Is this normal?
[image]
I noticed that there are:Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
Is that the reason? What do I need to do to correct the error?
[image]

Heritagecll, 2024-04-07:
It seems that we can enter the training phase normally.
[image]
But I can't see anything through TensorBoard, and I'm not sure if the current situation is correct.
[image]
Thank you very much for your sharing and help!

reiniscimurs, 2024-04-08:
You should be able to set any version of ROS you want in a virtual env. For virtual env problems, see other issues on the topic.