reiniscimurs/DRL-robot-navigation #131: Changing the lidar angel from 180 to 360 degrees will make the model to perform better or not?

am-amani, 2024-04-10:
Dear @reiniscimurs ,

One question. Have you tried or heard someone try to successfully train the mobile robot with lidar at a 360-degree angle instead of 180 degrees? In your code, you are using 180 degrees. I was wondering if changing it to 360 degrees will increase the trained agent's performance. I understand it will take longer to train the model, but I wondered if changing the environment dimension from 20 to 40 will help the robot perform better.

Thank you for your attention.

reiniscimurs, 2024-04-11:
Hi,

You can go through some still open and closed issues. Some people have tried this approach with varying results. There are papers out there that use 360 degree FOV but it is hard for me to say if that improves the performance.

am-amani, 2024-04-22:
Thank you for your reply!

closed 2024-04-22 by am-amani