reiniscimurs/DRL-robot-navigation #133: Change world file

Ethan0207, 2024-04-12:
Hi,
I just want to change the original TD3.world to model.world. And I referred to the issue #30  #64  #129 #16 . I **export GAZEBO_RESOURCE_PATH=~/DRL-robot-navigation-scout1/catkin_ws/src/scout/scout_gazebo_sim**. It just appeared an empty world and my robot.
[image]
[image]
Then I 
**export GAZEBO_RESOURCE_PATH=~/DRL-robot-navigation-scout1/catkin_ws/src/scout/scout_gazebo_sim/launch**.  It made an error about my robot.
[image]
1. What does resource path mean？
2. Could you give me some suggestions?
Thank you.

Ethan0207, 2024-04-29:
Hi,
Thank you for your reply. I  placed the 'tree.world' in the same directory where 'TD3.world' and using the same resource path as when using that world file. But it just appeared an empty world. And I used an another world file "lemon.world" and placed the  in the same directory where 'TD3.world' .But it was very strange. The world appeared, but my robot disappeared, it had some errors. Finally, I used "TD3.world", it is normal. I could not solve it and absolutely lost. Could you give me some suggestions?
[image]
[image]
[image]

reiniscimurs, 2024-04-30:
It is difficult for me to say from these images. In your [first co](https://github.com/reiniscimurs/DRL-robot-navigation/issues/133#issue-2239481304) I can see that your file structure is quite different from the default one in the repo so it would be very difficult for me to guess where the issues come in. But in the last images I can see a similar structure to the repo, so it is confusing to me what the final structure of your project is and as such I cannot guess if the paths are set up correctly.

However, in the last image your error seems unrelated to the map itself and the map seems to load so maybe you have set it correctly but there is an issue with the other map you are trying to load. In the last image your error comes from trying to use the robot model befor it is spawned in the scene. It might be that the map requires larger time span to load. You can try to increase the wait time in the code before trying to start training:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/train_velodyne_td3.py#L253

Ethan0207, 2024-04-30:
Hi，
Thank you for your help. When I increase the wait time to 30 seconds, it can load my model. This has been bothering me for a long time. Thank you again for your patient explanation.

closed 2024-05-01 by reiniscimurs

Seher-789, 2024-07-30:
@Ethan0207 how did you add delay?

Ethan0207, 2024-07-31:
here
https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/train_velodyne_td3.py#L253