reiniscimurs/DRL-robot-navigation #134: why my environment is empty

bb0928, 2024-04-12:
I followed the tutorial, when I launched train_velodyne_td3.py, RViz didn't produce any indoor environment, with error like:

[ERROR] [1712917359.424777513, 1.823000000]: Updating ModelState: model [cardboard_box_0] does not exist
[ERROR] [1712917359.424956752, 1.823000000]: Updating ModelState: model [cardboard_box_1] does not exist
[ERROR] [1712917359.425014172, 1.824000000]: Updating ModelState: model [cardboard_box_2] does not exist
[ERROR] [1712917359.425060643, 1.824000000]: Updating ModelState: model [cardboard_box_3] does not exist
[ERROR] [1712917438.270502562, 54.587000000]: Updating ModelState: model [cardboard_box_0] does not exist
[ERROR] [1712917438.270569909, 54.587000000]: Updating ModelState: model [cardboard_box_1] does not exist
[ERROR] [1712917438.270614372, 54.587000000]: Updating ModelState: model [cardboard_box_2] does not exist
[ERROR] [1712917438.270777024, 54.587000000]: Updating ModelState: model [cardboard_box_3] does not exist

what should I do? thanks a lot.

bb0928, 2024-04-12:
Sorry I forgot to modify my path.

closed 2024-04-14 by bb0928

reiniscimurs, 2025-08-11:
> Hi, I have the same problem. How did you fix it?

See any issue marked with resource path:
https://github.com/reiniscimurs/DRL-robot-navigation/issues?q=is%3Aissue%20state%3Aclosed%20label%3Aresouce_path