reiniscimurs/DRL-robot-navigation #137: When I run the training code, there will be an error

pihkhggc, 2024-04-24:
ROS Noetic on Ubuntu 20.04 with python 3.8.10 and pytorch 1.10.

error：
1.
Resource not found: multi_robot_scenario
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/opt/ros/noetic/share
The traceback for the exception was written to the log file

2.
RLException: roscore cannot run as another roscore/master is already running. 
Please kill other roscore/master processes before relaunching.
The ROS_MASTER_URI is http://gj-VirtualBox:11311/
The traceback for the exception was written to the log file

Although Gazebo and rviz seem to run smoothly without any impact, train_velodyne-td3. py cannot print out information

reiniscimurs, 2024-04-24:
Hi.

Have you followed the guide and sourced everything properly each time you start a new terminal. Also see the section on closing an already running ros and how to kill all the processes.

pihkhggc, 2024-04-25:
I followed the advice of others and wrote a launch file as follows
[image]

And annotated the part of the code in the velody_env.py file that opens the launch file
[image]

A terminal runs the launch file to start rviz and gazebo, while a new terminal runs train_velodyne-td3 .py, rviz and gazebo are running normally, but the following warnings appear and the training window will not print training information
[image]
[image]
[image]

Is it the problem caused by my modifications? Can you ask for your advice

pihkhggc, 2024-04-25:
Before I make any changes, follow your steps to start training train_velodyne-td3 .py, I found that rviz can start normally, but the terminal still shows a large warning and Gazebo does not start properly
[image]
[image]
[image]

reiniscimurs, 2024-04-25:
Sorry, I am not quite following what the issues are that you are facing. Are these still the issues in your first comment or new issues? Why do you need to make changes to the launch method?

> Gazebo does not start properly

What do you mean with Gazebo not starting properly? Gazebo GUI will not be launched by default in this repo. Gazebo is not supposed to appear.

The warnings you see are time differences between frames. This happens when your real time and ros time go out of sync. Most likely (since you are running in a VM) you do not have enough compute resources for ros and you get out of sync.

pihkhggc, 2024-04-30:
Thank you very much for your reply. It was very useful to me. I switched to a different computer and ran it again according to the steps. There was indeed no warning, but it got stuck in 【 urdf_spawner-2 】 and no other information was printed. Also, the car sometimes rotates in place. Are these all normal situations?
[image]
[image]

reiniscimurs, 2024-04-30:
This looks fine to me and seems that the model is training. You will have to wait for some time to see epochs information and for multiple epochs to see some more reasonable behavior.

Mehmet-Gozukucuk, 2024-05-27:
Hi there, I already trained my agent on my pc and tested it. Everything was okay and  right now I am trying to make it in to real world I have a raspberry pi 4 8GB ram and I did all the process ( seting up Ubuntu,  ROS, Pytorch). I load the model that I trained on my computer yesterday I tried to test model and I could see that rviz and gazebo run with no problem but today I tried to run test command again and I got the same error ;

Resource not found: multi_robot_scenario
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/opt/ros/noetic/share
The traceback for the exception was written to the log file 

There were no errors when I run test command on my own pc and when I run it on raspberry for the first time. I didn't get you you solved the problem. Any advices would be so appreciated, I am doing it for my graduation project as a 4th year of electrical and electronics engineer student.

Mehmet-Gozukucuk, 2024-05-27:
I have Ubuntu 20.04 and ROS noetic if you need this information.

reiniscimurs, 2024-05-27:
Most likely you have the same issue with wrong sourcing of ROS.

See the tutorial how to properly source files: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part1-installation-d62715722303

Or see any of these issues: https://github.com/reiniscimurs/DRL-robot-navigation/issues?q=is%3Aissue+is%3Aclosed+label%3Aresouce_path

Mehmet-Gozukucuk, 2024-05-27:
when I try command catkin_make_isolated, I also get this error

G2mobilrobot@raspberrypi:~/catkin_ws/src/DRL-robot-navigation/catkin_ws$ catkin_make_isolated 
Base path: /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws
Source space: /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/src
Build space: /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/build_isolated
Devel space: /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/devel_isolated
Install space: /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~
~~  traversing 4 packages in topological order:
~~  - multi_robot_scenario
~~  - velodyne_description
~~  - velodyne_gazebo_plugins
~~  - velodyne_simulator
~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing catkin package: 'multi_robot_scenario'
==> cmake /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario -DCATKIN_DEVEL_PREFIX=/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/devel_isolated/multi_robot_scenario -DCMAKE_INSTALL_PREFIX=/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/install_isolated -G Unix Makefiles in '/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/build_isolated/multi_robot_scenario'
CMake Error: The current CMakeCache.txt directory /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/build_isolated/multi_robot_scenario/CMakeCache.txt is different than the directory /home/mehmet/catkin_ws/src/DRL-robot-navigation/catkin_ws/build_isolated/multi_robot_scenario where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/CMakeLists.txt" does not match the source "/home/mehmet/catkin_ws/src/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
<== Failed to process package 'multi_robot_scenario': 
  Command '['cmake', '/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario', '-DCATKIN_DEVEL_PREFIX=/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/devel_isolated/multi_robot_scenario', '-DCMAKE_INSTALL_PREFIX=/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/build_isolated/multi_robot_scenario && cmake /home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario -DCATKIN_DEVEL_PREFIX=/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/devel_isolated/multi_robot_scenario -DCMAKE_INSTALL_PREFIX=/home/G2mobilrobot/catkin_ws/src/DRL-robot-navigation/catkin_ws/install_isolated -G 'Unix Makefiles'

Command failed, exiting.

reiniscimurs, 2024-05-27:
You can try using suggestion from here: https://github.com/reiniscimurs/DRL-robot-navigation/issues/99