reiniscimurs/DRL-robot-navigation #143: Problems in training

Sunsundayz1, 2024-05-14:
Hi, I'm training Gazebo environment. but I don't know where his target point is. Can you tell me please?

reiniscimurs, 2024-05-14:
Please describe the question in more detail.

Sunsundayz1, 2024-05-15:
[image]Hi, I'm not finding the target point in this training environment.
[image]
I don't know how to determine if target how to reach

reiniscimurs, 2024-05-15:
Target is not displayed in Gazebo environment. You can see it in Rviz though.

Setting and reaching target criteria are explained in the tutorial: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part4-environment-7e4bc672f590

Sunsundayz1, 2024-05-15:
I am a newbie and I don't know where I can see the target points and then please see my picture which shows no point cloud data. Please can you tell me what I should do. Secondly please do you have a pre-trained model? Thanks again.

Sunsundayz1, 2024-05-15:
I am a newbie and I don't know where I can see the target points and then please see my picture which shows no point cloud data. Please can you tell me what I should do. Secondly please do you have a pre-trained model? Thanks again.
[image]

reiniscimurs, 2024-05-15:
I suggest following some tutorials regarding ROS and familiarizing yourself with ROS as that will help you understand the implementation. It will be very difficult to explain any issues you encounter without some basic knowledge there and without knowing the differences between Gazebo and Rviz. This kind of data you are looking for is displayed in Rviz and not in Gazebo GUI.

Sunsundayz1, 2024-05-15:
[image]
I know that this is the gui for my rviz. but when I look at pointclound2 I get an error. Thank you very much for your patience!

reiniscimurs, 2024-05-15:
You are specifying the wrong topic. We are using velodyne lidar for pointcloud, not camera. Please familiarize yourself with the tutorial. This can help you clear up confusion you might have about the implementation.

Sunsundayz1, 2024-05-15:
Thank you very much, I am now able to visualize the training properly, but I have one more request. Can you give me a pre-trained model. Thanks again for your generous sharing!
[image]

reiniscimurs, 2024-05-16:
This paper does not implement the whole functionality of the paper, just the DRL navigation training part.

Parts of the code in the paper are in: https://github.com/reiniscimurs/GDAE