reiniscimurs/DRL-robot-navigation #149: About Image Sensors

nlnlnl1, 2024-07-01:
Hello, thank you very much for sharing your work. I would like to ask if the car is equipped with a depth camera that can detect the following depth images. If there is a depth camera, where should I obtain it? If not, how can I improve the car model?
I hope you can give me some advice. Thank you
[image]

reiniscimurs, 2024-07-01:
Hi,

Yes, there is depth camera available as a sensor that is available on the robot. You can use rviz to see available sensors by topic. If it can be used for semantic segmentation, I do not know. You will have to test it for yourself.

nlnlnl1, 2024-07-01:
Hello, I checked the topic on rviz and found the depth camera sensor, but it doesn't seem to be working. I checked the screen and it kept showing completely black
[image]
Then my terminal has the following information, it seems that the depth camera is not properly configured，When I used the command image_saver to save the topic camera/depth/imageraw, I got the following image, which doesn't seem to be the depth image I wanted.
[image]
Sorry, I am a beginner in ROS. I am not sure if it is a file configuration issue with the robot, so I would like to ask you about it

nlnlnl1, 2024-07-01:
Hello, I checked the topic on rviz and found the depth camera sensor, but it doesn't seem to be working. I checked the screen and it kept showing completely black
[image]
Then my terminal has the following information, it seems that the depth camera is not properly configured，
[image]

When I used the command image_saver to save the topic camera/depth/imageraw, I got the following image, which doesn't seem to be the depth image I wanted.
[image]
Sorry, I am a beginner in ROS. I am not sure if it is a file configuration issue with the robot, so I would like to ask you about it

nlnlnl1, 2024-07-01:
Hello, I checked the topic on rviz and found the depth camera sensor,
[image]

 but it doesn't seem to be working. I checked the screen and it kept showing completely black
[image]
Then my terminal has the following information, it seems that the depth camera is not properly configured，
[image]

When I used the command image_saver to save the topic camera/depth/imageraw, I got the following image, which doesn't seem to be the depth image I wanted.
[image]
Sorry, I am a beginner in ROS. I am not sure if it is a file configuration issue with the robot, so I would like to ask you about it

reiniscimurs, 2024-07-01:
The debug information looks fine. That is just an information output, not a warning or an error.

You should check what the raw information is from the camera topics with `rostopic echo`. I cannot look into this as I have no way of testing this, but you can try to see some information in ros tutorials: https://classic.gazebosim.org/tutorials/?tut=ros_depth_camera