reiniscimurs/DRL-robot-navigation #151: rviz and gazebo load an empy world environment

answsdu, 2024-07-02:
Hi Reinis, so sorry to bother you,

**Describe the bug**

when I run "**python3 test_velodyne_td3.py**",it turn to the following pictures,it seems like rviz and gazebo load an empty world environment.
[image]
[image]
And I have check my set of GAZEBO_RESOURCE_PATH following  comment#16,I think it's correct,but the program still can't work well.
[image]

the terminal send out the following error:
[image]

**Desktop (please complete the following information):**
 - OS:ubuntu 20.04
 - ROS version: noetic
 - Repository branch: main

Could u pls help me solve this problem?Thank u very much!

answsdu, 2024-07-02:
Hi Reinis, the program works well  after I rebuild the repo.But there are a new problem now,the model training is particularly slow,it only do 40 Epoch for 10 hours!Could u please tell me what can I do to improve the speed of training?Thank u very much!
[image]

reiniscimurs, 2024-07-02:
There isn't much you can do to increase training speed besides what is mentioned in the tutorial. However, here it seems your model is probably not training properly, and you might want to train again with a different seed. See issues labeled with `convergence` and see if you notice any similarities.

answsdu, 2024-07-03:
Hi,Reinis
I don't make any change to the **"train_velodne_td3.py"** and **"velodyne_env.py"** two files. After 89 epochs,the training seemed to work a little better like the following picture:
[image]
Thanks for ur answer and patience! I'm just starting to learn about DRL.And I will see the [convergence](https://github.com/reiniscimurs/DRL-robot-navigation/issues/126) .

reiniscimurs, 2024-07-03:
The training is somewhat random so even without changes it may fail to converge. Though it does look like the values are improving, the performance should also be improving.

innocence-cloud, 2024-07-10:
Hello, I had the same problem, rviz loaded an empty world environment, and the terminal error message was the same. I would like to ask you how to rebuild the repo. How to solve this problem. I will be very grateful.

Seher-789, 2024-07-30:
@answsdu hi, i want to know how did you solve the empty world problem?