reiniscimurs/DRL-robot-navigation #152: Running train_velodyne_td3.py errors

Jerryzrh, 2024-07-03:
2024-07-03 10:29:54.952219: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2024-07-03 10:29:55.117319: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-07-03 10:29:56.343452: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Traceback (most recent call last):
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorboard/compat/__init__.py", line 42, in tf
    from tensorboard.compat import notf  # noqa: F401
ImportError: cannot import name 'notf' from 'tensorboard.compat' (/home/z2823753580/.local/lib/python3.8/site-packages/tensorboard/compat/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/z2823753580/PycharmProjects/DRL-robot-navigation/TD3/train_velodyne_td3.py", line 9, in <module>
    from torch.utils.tensorboard import SummaryWriter
  File "/home/z2823753580/.local/lib/python3.8/site-packages/torch/utils/tensorboard/__init__.py", line 12, in <module>
    from .writer import FileWriter, SummaryWriter  # noqa: F401
  File "/home/z2823753580/.local/lib/python3.8/site-packages/torch/utils/tensorboard/writer.py", line 16, in <module>
    from ._embedding import get_embedding_info, make_mat, make_sprite, make_tsv, write_pbtxt
  File "/home/z2823753580/.local/lib/python3.8/site-packages/torch/utils/tensorboard/_embedding.py", line 9, in <module>
    _HAS_GFILE_JOIN = hasattr(tf.io.gfile, "join")
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorboard/lazy.py", line 65, in __getattr__
    return getattr(load_once(self), attr_name)
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorboard/lazy.py", line 97, in wrapper
    cache[arg] = f(arg)
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorboard/lazy.py", line 50, in load_once
    module = load_fn()
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorboard/compat/__init__.py", line 45, in tf
    import tensorflow
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/__init__.py", line 38, in <module>
    from tensorflow.python.tools import module_util as _module_util
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/__init__.py", line 45, in <module>
    from tensorflow.python.feature_column import feature_column_lib as feature_column
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/feature_column/feature_column_lib.py", line 18, in <module>
    from tensorflow.python.feature_column.feature_column import *
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/feature_column/feature_column.py", line 143, in <module>
    from tensorflow.python.layers import base
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/layers/base.py", line 16, in <module>
    from tensorflow.python.keras.legacy_tf_layers import base
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/keras/__init__.py", line 25, in <module>
    from tensorflow.python.keras import models
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/keras/models.py", line 25, in <module>
    from tensorflow.python.keras.engine import training_v1
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/keras/engine/training_v1.py", line 46, in <module>
    from tensorflow.python.keras.engine import training_arrays_v1
  File "/home/z2823753580/.local/lib/python3.8/site-packages/tensorflow/python/keras/engine/training_arrays_v1.py", line 37, in <module>
    from scipy.sparse import issparse  # pylint: disable=g-import-not-at-top
  File "/usr/lib/python3/dist-packages/scipy/sparse/__init__.py", line 229, in <module>
    from .base import *
  File "/usr/lib/python3/dist-packages/scipy/sparse/base.py", line 8, in <module>
    from .sputils import (isdense, isscalarlike, isintlike,
  File "/usr/lib/python3/dist-packages/scipy/sparse/sputils.py", line 16, in <module>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/usr/lib/python3/dist-packages/scipy/sparse/sputils.py", line 16, in <listcomp>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/home/z2823753580/.local/lib/python3.8/site-packages/numpy/__init__.py", line 320, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'

reiniscimurs, 2024-07-03:
Hi,
Please provide information according to the issue template. Simple error message gives no context to your problem.

Jerryzrh, 2024-07-05:
[image]

Jerryzrh, 2024-07-05:
z2823753580@ubuntu:~/DRL-robot-navigation/TD3$ python3 train_velodyne_td3.py
Roscore launched!
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
... logging to /home/z2823753580/.ros/log/b9e10b84-3a95-11ef-8986-3312b264d3db/roslaunch-ubuntu-18070.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:32777/
ros_comm version 1.16.0

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES

auto-starting new master
process[master]: started with pid [18080]
Gazebo launched!
ROS_MASTER_URI=http://localhost:11311/

setting /run_id to b9e10b84-3a95-11ef-8986-3312b264d3db
process[rosout-1]: started with pid [18093]
started core service [/rosout]
... logging to /home/z2823753580/.ros/log/b9e10b84-3a95-11ef-8986-3312b264d3db/roslaunch-ubuntu-18088.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: multi_robot_scenario
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/opt/ros/noetic/share
The traceback for the exception was written to the log file