reiniscimurs/DRL-robot-navigation #153: Q?

AtmAbdelkader, 2024-07-03:
Hi this a great project thankyou to share it , i transfert the code to ros2 humble and  is working , now i change the algorithm to PPO but is not working can you give me some tips and tricks to implement this algorithm without problems

reiniscimurs, 2024-07-03:
Hi,

Seems like you are implementing a different backend and an algorithm. I don't see how I could be of help there, sorry.

AtmAbdelkader, 2024-07-27:
Hello sir!
How can visualize the goal point in gazebo?

reiniscimurs, 2024-07-28:
The goal is visualized in rviz. Visualization in gazebo is not supported here

AtmAbdelkader, 2024-07-28:
ok, thankyou

AtmAbdelkader, 2024-07-29:
Hello, 

i think rviz have a 3d visualization how did you make 2d visualization, Is with velodyne subscriber or another method ?