reiniscimurs/DRL-robot-navigation #158: different world issue

Seher-789, 2024-07-30:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - ROS version:
 - Repository branch:

**Additional context**
Add any other context about the problem here.

reiniscimurs, 2024-07-30:
Hi,

Please use the template to submit the issue.

1. Have you checked that you have set up the sources correctly? What does the log output say?
2. Can you load your map in gazebo?

Seher-789, 2024-08-03:
[image]
thank you for help. I did it by increase the time but now the problem is whenever i uplad the new world then i get this errors

reiniscimurs, 2024-08-05:
The code expects there to be the cardboard boxes in the world to reset them to random positions. You can simply update the code, not to reset the objects that are not a partof your world file. Random box reset is explained in the tutorial: https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part4-environment-7e4bc672f590

Seher-789, 2024-08-06:
thank you so much. for your help i did it.

closed 2024-08-06 by Seher-789