reiniscimurs/DRL-robot-navigation #159: use another robot

Seher-789, 2024-08-06:
hi, thank you so much for your help i have read the article on turtlebot3 but acutally my robot is customized robot and it is two differential driver robot so i want to spawn my customized robot in it i have made the changes so that i launch the robot but still i am facing errors can you  please help me in this or any guidelines. i also added my robot so if you guide me 
[image]

[image].
[image]

reiniscimurs, 2024-08-09:
Hi,

Please provide description of problems you are facing. It is difficult to gauge what issue is at hand here. Also, please note that I can only support within the relevant confines of this repo and it is given as is. I would probably not know of any individual extra changes you make to the repo or how to solve them.

1903943257, 2025-04-16:
> hi, thank you so much for your help i have read the article on turtlebot3 but acutally my robot is customized robot and it is two differential driver robot so i want to spawn my customized robot in it i have made the changes so that i launch the robot but still i am facing errors can you please help me in this or any guidelines. i also added my robot so if you guide me[Screenshot from 2024-08-06 17-52-27](https://github.com/user-attachments/assets/e1f1208f-27ed-47ee-a12d-0bd73d941883)
> 
> [image].  [ [image] ](https://private-user-images.githubusercontent.com/58245291/355401847-bd2ccf76-63ef-4a59-9a1c-540499835ba8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQ4MzgzOTUsIm5iZiI6MTc0NDgzODA5NSwicGF0aCI6Ii81ODI0NTI5MS8zNTU0MDE4NDctYmQyY2NmNzYtNjNlZi00YTU5LTlhMWMtNTQwNDk5ODM1YmE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDE2VDIxMTQ1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFkMWMxNDA0NGZkYTE0NTNkZWMyMjYyOTJjNmMzNGM0Yjc0MTFjMWJkZGVhODM0NzFiNTc0YzcxNmYzNzQzMzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.MVGgzGNz9xcBSnCsnSCRJ0QdTP0grx1vbEcirccOfHY)。 [image]

Hello, may I ask if you have succeeded? I am also working on a similar project and I want to get Unitree go2 running. I am not sure how to do it now. Can you give me some advice? Thank you very much!