reiniscimurs/DRL-robot-navigation #161: Sir, could you provide a trained model? please

zcjiyouxiusa, 2024-10-24:
hello, sir！I have some trouble with training.
why it is hard to train a well model with you origin code. 
I have run the `train_velodyne_td3.py` for a half of day with nvidia GPU: RTX 3060.  but the robot still can not navigate to its goal.
can you tell me how long to run to get a well model.
the following pic is the loss and Q value, The loss fluctuates greatly. Is this normal?
[image]
[image]
Please help me ! Thanks !

reiniscimurs, 2024-10-24:
Hi, please use issue template to submit your issue.

LSWings, 2025-05-19:
> hello, sir！I have some trouble with training. why it is hard to train a well model with you origin code. I have run the `train_velodyne_td3.py` for a half of day with nvidia GPU: RTX 3060. but the [...]

Hi，Can you tell me how you drew this data graph?

zcjiyouxiusa, 2025-05-20:
> > hello, sir！I have some trouble with training. why it is hard to train a well model with you origin code. I have run the `train_velodyne_td3.py` for a half of day with nvidia GPU: RTX 3060. but the robot still can not navigate to its goal. can you tell me how long to run to get a well model. the following pic is the loss and Q value, The loss fluctuates greatly. Is this normal? [image] ![Snipaste_2024-10-24_14-27-46](https://private-user-images.githubusercontent.com/63300271/379642592-937ccf80-3f44-42d4-92b5-f660872beb91.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDc3MTMxNDYsIm5iZiI6MTc0NzcxMjg0NiwicGF0aCI6Ii82MzMwMDI3MS8zNzk2NDI1OTItOTM3Y2NmODAtM2Y0NC00MmQ0LTkyYjUtZjY2MDg3MmJlYjkxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTIwVDAzNDcyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxYmYxMTg5MTE0ZTU0MDg0OTNlZDA4YWE4ODEzNzJkNTAwZGQ0NWM0MzQwZmZlMTBhZTAzZGU3MWQxNGNkMDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.S_CzeFIQtbrRr8O-PSAZcSep7gVH4H1oik6KuhvWyI8) Please help me ! Thanks !
> 
> Hi，Can you tell me how you drew this data graph?

>Use tensorboard
```
$ cd ~/DRL-robot-navigation/TD3
$ tensorboard --logdir runs
```