reiniscimurs/DRL-robot-navigation #162: torch.Size does not match expected state_dim

xjuzyw0319, 2024-10-28:
**Describe the issue**
I am going to improve the replay_buffer in your source code and add the PrioritizedReplayBuffer model. I am currently facing a dimension mismatch problem. The torch.Size dimension printed each time I run the code is inconsistent. I originally analyzed that it was caused by the random start and target points. The following figure 1 is the result of my printing, and the other pictures are some of my modifications to the source code. Please give me some suggestions, thank you very much.

**Screenshots**
[image]图一
[image]
[image]
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

reiniscimurs, 2024-10-28:
Hi,
At some point your code has a mistake in how it returns a state from your replay buffer. Again, it will not be the start and goal points, but rather some return from the buffer or else you would never be able to add any experiences to it.

However, you have not shared all the changes you have made and without that I cannot help you here as it is not possible for me to guess how you have implemented sampling. 

Good luck

xjuzyw0319, 2024-10-31:
I rewrote your experience pool (as shown in Figure 1), but during the training process, I found that after a few rounds of training, the agent kept spinning in place（as shown in Figure 2） and the success rate was always 0 (as shown in Figure 3). Could you give me some suggestions?
[image]
图1
[image]
图2
[image]图3