reiniscimurs/DRL-robot-navigation #163: Robot moving in circle after set the multiple seed and change the Lr

Seher-789, 2024-10-31:
i have trained my robot many time and i changed the seed from 0 to 42 but no result even change the lr from 1e-3 to 1e-4 but no result please suggest me some solution
[image]

reiniscimurs, 2024-10-31:
Hi,

Please use the submission template when submitting the issue. This gives me the necessary information to start figuring out what the issue is.

How many epochs did you try training with each seed?

Seher-789, 2024-11-01:
Hi, Thank you for response i trained for 75 but the reward was negative.

reiniscimurs, 2024-11-01:
Again, please provide full information as in the template.

75 **epochs** for each seed? Seems excessive. If it does not converge at such time for any seed there is either some change or mistake in the implementation. Here, the template information is needed.

Seher-789, 2024-11-06:
thank you for response , No i have trained the robot 75 epochs for seed 2 and seed 0 but the rewards were negative and i have trained with all the other seed values for ten epochs but the rewards were negative after the validating start the robot starts to move in circle

reiniscimurs, 2024-11-06:
One last time, provide the information as set in the issue template. It is always very difficult to answer questions without knowing the full information.

TD3 can be a bit iffy with training and could also not converge but if no changes were made to the code I would expect it to work with at least some random seed. 10 epochs might still not be enough though to evaluate the performance. But if longer training does not help, you could try to use different learning rates.

bilalabidbuttisking, 2025-03-12:
Hello, I am using exactly your code without any changes and training has started. Right now I am at EPOCH 144. But the it has not shown any improvement. Is it normal?

reiniscimurs, 2025-03-12:
> Hello, I am using exactly your code without any changes and training has started. Right now I am at EPOCH 144. But the it has not shown any improvement. Is it normal?

144 epochs would mean something like 2 or 3 days of training. Are you sure it has trained for that long?
In any case some sort of behavior would usually start at about 20 to 40 epochs so you could just restart the training with a different seed value.