reiniscimurs/DRL-robot-navigation #169: How to determine if training can be stopped？

qwsdxcdd, 2025-02-08:
**Describe the issue**
Dear Sir, hello!This is my first time training in deep reinforcement learning and reproducing your project. I would like to ask how to determine if my training can be terminated. Currently, robots can reach their target points in the environment during training, but there are also times when they encounter obstacles between very narrow obstacles and have to restart. I am not sure if this is a normal phenomenon. I have been training for 21 hours and have reached 95 epochs. Here are my training images for your review. I also hope you can give me advice on whether to stop training or continue training for more epochs.Looking forward to your reply, I would greatly appreciate it!

**Screenshots**

[image]

[image]

[image]

qwsdxcdd, 2025-02-08:
[image]

reiniscimurs, 2025-02-08:
There is no one right answer for this, but generally, when your AvQ and MaxQ curves have converged that is a good sign that the training can be stopped. But I would also evaluate the qualitative results in the evaluation stage.