reiniscimurs/DRL-robot-navigation #170: Shortest path planning

qwsdxcdd, 2025-02-10:
**Describe the issue**
Hello, I have successfully trained and tested your project, basically you can reach the target point, but the path is not the shortest (optimal), I would like to ask if it is feasible to combine your reinforcement learning strategy with traditional path planning algorithms to reach the shortest path, what are your suggestions? Looking forward to your reply, I would appreciate it

reiniscimurs, 2025-02-10:
Hi, 
Surely you can try to. But this task would be somewhat ill-formed. Path planners work with some information about the environment whereas RL here takes only single state information. It can not be globally optimal since it works with only local time and space information. You can implement some sort of a combination of both as was done in GDAE and similar works where planner calculates the global plan and RL policy takes care of local motion planning. This has proven to work well. Other methods would be entirely up to you.

sattlelite, 2025-02-19:
I totally agree with that. Since the approach based on a planner like SLAM often needs global map information to calculate the optimal path. So any optimal path planning methods would not work until you got the global map. Is it right?

reiniscimurs, 2025-02-25:
That is true. Globally optimal plans need full information of the environment (A* for example). If that is missing, you cannot be certain if it is globally optimal.

qwsdxcdd, 2025-04-26:
> That is true. Globally optimal plans need full information of the environment (A* for example). If that is missing, you cannot be certain if it is globally optimal.

Hello, please excuse me, after training and testing in this repo, each random starting point can basically reach a random target point, can this project be used as a global path planning from a random starting point to a random target point?
In addition, if according to the above statement, this repo can be regarded as global path planning, then, reinforcement learning is local path planning, what is the training strategy that allows the robot to reach the target point step by step? There is a POI in GDAE to guide the robot to the target point, but this repo does not involve a POI, what guides the robot to the target point? How should I understand it, is it a reward function? Or what, I'm looking forward to your answer. Thank you for having a responsible worker like you, it's appreciated.

sattlelite, 2025-04-26:
Hello,
I agree with your point. Reiniscimurs is a very responsible researcher.
Regarding your question, as I understand it. Unlike the planning-based approach, this reinforcement learning method is designed to train a decision-making network to make the optimal action in each state. The state space serves as the input to this network, and the decision action at each moment is the output. The reward function can be regarded as the training task set for this decision-making network. As David_Silver mentioned, reinforcement learning is an independent machine learning paradigm aimed at maximizing the reward function. Therefore, the robot gradually learns the behavior to reach the goal from random exploration actions. This is the basic concept of reinforcement learning.