reiniscimurs/DRL-robot-navigation #171: Unable to load obstacle model

lhqll, 2025-03-04:
Hello, dear Reinis Cimurs,
Thank you for publishing such an important research result. I want to apply this model to our own research.
Here are some problems I have encountered while running the simulation. When I ran this model, an empty map appeared and the robot was running outside the simulation map. As I am just beginning to learn ROS system, I hope to get your help.

[image]

[image]

[image]

Based on your suggestion, I have tried the following solutions.

[image]

[image]

[image]

When I do the first time "echo $GAZEBO_RESOURCE_PATH" then it shows empty space, after exporting all the things then it shows us the above source path. I am also not so experienced with ROS and Gazebo, so for this reason maybe for me little troublesome to identify a problem.

**Preliminaries**
 - Have you checked the tutorial for answers to your issue?  Yes
 - Have you checked prior issues for answer to your issue?  Yes

**Provide information about your setup:**
 - OS: ubuntu 20.04
 - ROS version: noetic
 - Repository branch: main

reiniscimurs, 2025-03-04:
Hi,

By the looks of the final image, the resource path is set up in your execution. The issue here is that there is no expected world file in that location so an empty world is loaded instead. What is the full path that you cloned the repo to? Is the path you are exporting the correct one? Can you follow this path in your system and fine the world file there?

lhqll, 2025-03-04:
> Hi,
>
> By the looks of the final image, the resource path is set up in your execution. The issue here is that there is no expected world file in that location so an empty world is loaded instead. What [...]

Thank you for your reply, My full path is /home/lhq1/catkin_ws/src/DRL-robot-navigation. I can follow the path to the world file in my system. 

[image]

[image]

reiniscimurs, 2025-03-04:
Ok I see what the issue might be. There are 2 points here. First, the exported value (i.e. GAZEBO_RESOURCE_PATH) is only valid in the terminal that you exported it in. So if you open a new terminal and run `echo $GAZEBO_RESOURCE_PATH` there it will be nothing to show there.

Second, try exporting `export GAZEBO_RESOURCE_PATH=~/catkin_ws/src/DRL-robot-navigation/catkin_ws/src/multi_robot_scenario/launch`
You are missing a / before your path so it looks like it is not resolved correctly.

lhqll, 2025-03-04:
> Ok I see what the issue might be. There are 2 points here. First, the exported value (i.e. GAZEBO_RESOURCE_PATH) is only valid in the terminal that you exported it in. So if you open a new terminal [...]

Thank you very much for your help, I have successfully run this model.

[image]