reiniscimurs/DRL-robot-navigation #172: ModuleNotFoundError: No module named 'squaternion'

sugar-Steamedfish, 2025-03-11:
**Describe the issue**
When I reproduce your code, I run python train_velodyne_td3.py, which reminds me that No module named 'squaternion', but I have looked up for a long time and still don't know where to download the package. This import is known as line 15 of velodyne_env.py: "from squaternion import Quaternion"

**To Reproduce**

python train_velodyne_td3.py

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Log output**

[image]

**Preliminaries**
 - Have you checked the tutorial for answers to your issue?
 - Have you checked prior issues for answer to your issue?

**Provide information about your setup:**
 - OS:ubuntu18.04
 - ROS version:melodic
 - Repository branch:main

**Additional context**
Add any other context about the problem here.

reiniscimurs, 2025-03-11:
[pip squaternion](https://pypi.org/project/squaternion/)
`pip install squaternion`

You can use any Quaternion package here that you wish but you would need to make sure the same input structure is used.

sugar-Steamedfish, 2025-03-11:
> [PIP 四元数](https://pypi.org/project/squaternion/) `pip install squaternion`
> 
> 您可以在此处使用任何您想要的 Quaternion 包，但您需要确保相同的输入结构被使用。
Thank you for your reply. In fact, I have done this operation before, and I did not know why I could not find the package at that time, but I succeeded this time.
In addition, when I am training, there is no visual log output to help me judge.
[image]

reiniscimurs, 2025-03-11:
What visual log output is expected? Looks like it runs as intended for me.

Rviz should be launched automatically. Gazebo simulator you would have to enable. Any other information will be posted in terminal, but by default the only information in terminal is only during evaluation runs and not during training. If you want to see training progress, you would have to look at tensorboard.

sugar-Steamedfish, 2025-03-11:
> 预期会有什么样的可视化日志输出？看起来它按照我的预期运行。
> 
> Rviz 应该会自动启动。您必须启用的 Gazebo 模拟器。任何其他信息都将发布在终端中，但默认情况下，终端中的唯一信息仅在评估运行期间发布，而不是在训练期间。如果您想查看训练进度，则必须查看 tensorboard。

Thank you for your reply and wish you a happy life

closed 2025-03-11 by sugar-Steamedfish