reiniscimurs/DRL-robot-navigation #175: How to add a obstacle? like a cube

dischabo, 2025-03-22:
Thank you for such a great project first
As the title, is it in the velodyne_env.py? I tried but failed.
Maybe it is difficult? Sorry, I don't learn about this much as a beginner.

reiniscimurs, 2025-03-22:
Hi,

Obstacles are defined in the world file: https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/catkin_ws/src/multi_robot_scenario/launch/TD3.world

`world` files can be opened in gazebo and directly edited there.

World file that is used in this repo is defined here:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/aba927a6a3b832a15e02183a2a9c78fce7a8732e/catkin_ws/src/multi_robot_scenario/launch/empty_world.launch#L16

dischabo, 2025-03-22:
Thank you very much for the answer!

closed 2025-03-22 by reiniscimurs