reiniscimurs/DRL-robot-navigation #176: Training issue: Robot moves in circles and fails to reach goal after 1000 episodes

mxjily, 2025-04-03:
Hi, thank you for sharing this work!

I followed the training process described in the paper and code, but after training for more than 3000 episodes, I found that the robot still cannot reach the goal. Instead, it keeps moving in a circular trajectory.

Do you have any suggestions on how to resolve this issue? Could it be related to hyperparameters, reward settings, or environment configuration?

Any help would be appreciated. Thanks again!

reiniscimurs, 2025-04-03:
Hi,

How many epochs did you train for? Generally, you should see some more reasonable behavior after 20 to 40 epochs. This is not a quick process so it can take a while.

Alternatively you can think of adapting the code and hyperparameters from https://github.com/reiniscimurs/DRL-robot-navigation-IR-SIM as that is basically the same code implementation with some added tuning.
At least the Kaiming initialization of network weights. I find that to stabilize the training
https://github.com/reiniscimurs/DRL-robot-navigation-IR-SIM/blob/fdcaa8dad68d2950513c276f292993dd29fb5a1f/robot_nav/models/TD3/TD3.py#L34

mxjily, 2025-04-04:
Hi,

First of all, thank you very much for your suggestions and help in solving my previous problem.

I was using the Noetic-Turtlebot branch and compiled the project with catkin_make. After training for around 17 epochs, I observed the robot spinning in place and failing to reach the goal.

Yesterday, I switched to the main branch and compiled it using catkin_make_isolated. Now, after training for 44 epochs, the robot is able to reach the goal. However, it still occasionally spins around the goal instead of stopping.

I also noticed that the loss is increasing during training. I have attached a chart showing the trend of the loss, average Q, and max Q values during training — the blue curve is from the Noetic-Turtlebot branch and the orange curve is from the main branch.

Could you please help me analyze if this behavior and these trends are expected?

Additionally, I recorded a short video showing the robot's behavior after training for 44 epochs on the main branch.

Thanks again for your time and helpful advice. I’ll definitely try the recommendations you mentioned for future training.

Best regards,

[image]

[image]

[image]
https://youtu.be/CY0vcC2J8HE

reiniscimurs, 2025-04-04:
Hi,

I wouldn't expect that the compilation has any effect besides just starting the training again with maybe different initialization of weights. The training is somewhat random and the random initialization can have a huge impact. This is why I would suggest adding Kaiming initialization.

The loss curves look fine. Since it is not an imitation learning problem there is no expectation for the loss to constantly decrease as we do not have ground truth with which to compare and optimize the model over. Rather, loss is a comparison of target and current q value, both of which are just an estimation of what the q value might be and will be updated. It might mean that the target q value is lagging behind and is an out of date q estimation policy, so you might want to experiment with the tau parameter to add more of the current policy in the target policy when using the soft update.

For the rotation, it just hasn't figured out the policy yet. You can see your average and max Q values increasing, this means that the policy is learning actions that it thinks are good. However, they have not yet converged so the policy should still be trained quite a bit more.

mxjily, 2025-04-05:
Hi @reiniscimurs ,

After more than two days of training (158 epochs), our model’s loss has stabilized around 140. Based on our analysis, improper weight initialization could lead to gradient vanishing or explosion — but in such cases, the loss wouldn’t stabilize. Therefore, we believe our training is still in a healthy regime. That said, applying Kaiming initialization could help the network converge faster and more reliably, which we plan to test soon.

We also examined the effect of the tau parameter. A very small tau could indeed make the loss larger in early training due to slow updates of the target network. However, we observed that even after the average Q value (AvQ) stabilized, the loss didn't decrease further — this suggests tau primarily impacts early-stage learning. A more appropriate tau might help loss settle faster but may not lower it significantly in the long run.

Despite the loss stabilizing around 140, the AvQ trend shows the critic network has learned meaningful estimations. In RViz, the robot can now successfully navigate to the goal point, which further supports that learning is taking place.

We also suspect that the loss can’t go much lower due to two factors:

The noise added to actions during target Q computation, and

Reward inconsistency in visually or geometrically similar scenarios (e.g., small variations in laser readings may lead to very different rewards).

That said, we’ve noticed two issues during navigation:

The robot tends to jitter or pause when the goal is very close to an obstacle.

In large L-shaped wall corners (when the robot is inside and the goal is just outside near the corner), the robot struggles or completely fails to reach the goal.

Could you kindly let us know if this analysis makes sense to you? Also, based on your previous advice, we plan to experiment with different tau values and use Kaiming initialization. We’ll also look into modifying the reward function to handle these specific edge cases.

Once again, thank you for your guidance! Below are the training curves for 158 epochs showing AvQ, MaxQ, and Loss.

[image]

[image]

[image]

reiniscimurs, 2025-04-05:
Hi,

There should not be vanishing gradient as the network is not deep enough and does not really use any sigmoid or tanh activations besides the last layer. The AvQ value seems a bit too low for me, I would expect it to settle around 60 or so. So while it does learn the proper max Q values then average Q value seems a bit understated so the learned policy might not be optimal.

I don't think the point about the laser values not changing much over steps but producing different rewards holds as reward is direct result of the state which includes all the parameters - laser state, motion and goal point. There might be an argument that it is too granular, but in this implementation and others this works well in practice and reduces the complexity of the model. A [1D cnn](https://github.com/reiniscimurs/DRL-robot-navigation-IR-SIM/blob/master/robot_nav/models/CNNTD3/CNNTD3.py) implementation over laser states does work better as the model has a better chance of learning more impactful geometric features from laser, but this will not make a huge impact.

For reaching goals, there is no local memory for robot and each state is treated individually. So once robot reaches some sort of a local optimum, it will not be able to escape it simply because it does not know that it is stuck in a loop. For this some [RNN](https://github.com/reiniscimurs/DRL-robot-navigation-IR-SIM/blob/master/robot_nav/models/RCPG/RCPG.py) implementation can be used.

Generally, I do not see any issues here that would not be expected. Training here is somewhat unstable and it is stochastic so you will get a different performance every training cycle. You simply should train multiple times and use a policy that performs best.

mxjily, 2025-04-06:
Hi, @reiniscimurs 

First of all, thank you for the detailed explanation regarding vanishing gradients and why the reward won’t fluctuate unexpectedly — your clarification about the influence of the full state (laser, motion, and goal) was very helpful.

We’ll first try adjusting the initialization (e.g., using Kaiming init) and tuning the tau parameter to see if we can get the average Q value (AvQ) to rise closer to the expected 60.

Thanks again for your continuous support and guidance!

w4646, 2025-04-07:
> Hi, thank you for sharing this work!
>
> I followed the training process described in the paper and code, but after training for more than 3000 episodes, I found that the robot still cannot reach the [...]

Hello, can this project be run on an Ubuntu 22.04 system?

reiniscimurs, 2025-04-07:
> Hello, can this project be run on an Ubuntu 22.04 system?

Please don't hijack issues. Open a new issue according to the issue template.

ROS Noetic is not supported for Ubuntu 22.04 so it will not work straight out of box.

mxjily, 2025-04-07:
@w4646 You can try  running this project in docker, it may work well in ubuntu22.04.

w4646, 2025-04-08:
> Hi, thank you for sharing this work!
>
> I followed the training process described in the paper and code, but after training for more than 3000 episodes, I found that the robot still cannot reach the [...]

Hello, does the Gazebo environment in this project need to be built by oneself

mxjily, 2025-04-08:
@w4646 If you're running the project inside the Docker container, then you don't need to build the Gazebo environment yourself.

However, if you're running it on a native Ubuntu 22.04 system, then you do need to set up the Gazebo environment manually (including ROS, Gazebo plugins, launch files, etc.).

LSWings, 2025-05-19:
> [@reiniscimurs](https://github.com/reiniscimurs)
> 
> Hi,
> 
> First of all, thank you very much for your suggestions and help in solving my previous problem.
> 
> I was using the Noetic-Turtlebot branch and compiled the project with catkin_make. After training for around 17 epochs, I observed the robot spinning in place and failing to reach the goal.
> 
> Yesterday, I switched to the main branch and compiled it using catkin_make_isolated. Now, after training for 44 epochs, the robot is able to reach the goal. However, it still occasionally spins around the goal instead of stopping.
> 
> I also noticed that the loss is increasing during training. I have attached a chart showing the trend of the loss, average Q, and max Q values during training — the blue curve is from the Noetic-Turtlebot branch and the orange curve is from the main branch.
> 
> Could you please help me analyze if this behavior and these trends are expected?
> 
> Additionally, I recorded a short video showing the robot's behavior after training for 44 epochs on the main branch.
> 
> Thanks again for your time and helpful advice. I’ll definitely try the recommendations you mentioned for future training.
> 
> Best regards,
> 
> [image]
> 
> [image]
> 
> [image]
> 
> The video in this url: https://youtu.be/CY0vcC2J8HE

Hi，Can you tell me how you drew this data graph?