reiniscimurs/DRL-robot-navigation #177: The data does not seem to be recorded, and it keeps reminding redundant timestamps and unknown_publisher

Shepherd010, 2025-04-10:
**Describe the issue**
The data does not seem to be recorded. It keeps reminding redundant timestamps and unknown_publisher, and never outputs "Average Reward over %i Evaluation Episodes, Epoch %i: %f, %f", but the data in tensorboard is normal.

**To Reproduce**
Everything was installed according to the official tutorial, python=3.8, cuda=111, torch=1.10

**Expected behavior**
[ WARN] [1744263185.005520537, 276.412000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link (parent odom) at time 276.388000 according to authority unknown_publisher

**Screenshots**

[image]

**Log output**
[ INFO] [1744262698.340411078, 0.201000000]: DiffDrive(ns = r1/): <tf_prefix> =
[DEBUG] [1744262698.340533849, 0.201000000]: DiffDrive(ns = r1/): <commandTopic> = cmd_vel
[DEBUG] [1744262698.340596223, 0.201000000]: DiffDrive(ns = r1/): <odometryTopic> = odom
[DEBUG] [1744262698.341171908, 0.201000000]: DiffDrive(ns = r1/): <odometryFrame> = odom
[DEBUG] [1744262698.341209957, 0.201000000]: DiffDrive(ns = r1/): <robotBaseFrame> = base_link
[DEBUG] [1744262698.341294272, 0.201000000]: DiffDrive(ns = r1/): <publishWheelTF> = false
[ WARN] [1744262698.341346826, 0.201000000]: DiffDrive(ns = r1/): missing <publishOdomTF> default is true
[DEBUG] [1744262698.341411341, 0.201000000]: DiffDrive(ns = r1/): <publishWheelJointState> = true
[DEBUG] [1744262698.341503528, 0.201000000]: DiffDrive(ns = r1/): <wheelSeparation> = 0.29999999999999999
[DEBUG] [1744262698.341574385, 0.201000000]: DiffDrive(ns = r1/): <wheelDiameter> = 0.17999999999999999
[DEBUG] [1744262698.341636054, 0.201000000]: DiffDrive(ns = r1/): <wheelAcceleration> = 1.8
[DEBUG] [1744262698.341704023, 0.201000000]: DiffDrive(ns = r1/): <wheelTorque> = 20
[DEBUG] [1744262698.341772356, 0.201000000]: DiffDrive(ns = r1/): <updateRate> = 50
[DEBUG] [1744262698.341895255, 0.201000000]: DiffDrive(ns = r1/): <odometrySource> = world := 1
[DEBUG] [1744262698.341993786, 0.201000000]: DiffDrive(ns = r1/): <leftJoint> = left_hub_joint
[DEBUG] [1744262698.342200122, 0.201000000]: DiffDrive(ns = r1/): <rightJoint> = right_hub_joint
[ WARN] [1744262698.342308975, 0.201000000]: GazeboRosDiffDrive Plugin (ns = ) missing <publishTf>, defaults to 1
[ INFO] [1744262698.343031188, 0.201000000]: DiffDrive(ns = r1/): Advertise joint_states
[ INFO] [1744262698.343531662, 0.201000000]: DiffDrive(ns = r1/): Try to subscribe to cmd_vel
[ INFO] [1744262698.363106434, 0.201000000]: DiffDrive(ns = r1/): Subscribe to cmd_vel
[ INFO] [1744262698.364317039, 0.201000000]: DiffDrive(ns = r1/): Advertise odom on odom
[ INFO] [1744262698.369069419, 0.201000000]: GazeboRosJointStatePublisher is going to publish joint: chassis_swivel_joint
[ INFO] [1744262698.369152000, 0.201000000]: GazeboRosJointStatePublisher is going to publish joint: swivel_wheel_joint
[ INFO] [1744262698.369190659, 0.201000000]: GazeboRosJointStatePublisher is going to publish joint: left_hub_joint
[ INFO] [1744262698.369240561, 0.201000000]: GazeboRosJointStatePublisher is going to publish joint: right_hub_joint
[ INFO] [1744262698.369285510, 0.201000000]: Starting GazeboRosJointStatePublisher Plugin (ns = r1/)!, parent name: r1
[DEBUG] [1744262698.380965973, 0.201000000]: Trying to publish message of type [sensor_msgs/LaserScan/90c7ef2dc6895d81024acba2ac42f369] on a publisher with type [sensor_msgs/LaserScan/90c7ef2dc6895d81024acba2ac42f369]
[DEBUG] [1744262698.394830764, 0.201000000]: Trying to publish message of type [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7] on a publisher with type [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7]
[DEBUG] [1744262698.395028360, 0.201000000]: Trying to publish message of type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd] on a publisher with type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd]
[DEBUG] [1744262698.691218427, 0.470000000]: Trying to publish message of type [sensor_msgs/CameraInfo/c9a58c1b0b154e0e6da7578cb991d214] on a publisher with type [sensor_msgs/CameraInfo/c9a58c1b0b154e0e6da7578cb991d214]
[urdf_spawner-3] process has finished cleanly
log file: /root/.ros/log/9da1104c-15cb-11f0-94cb-0242ac110002/urdf_spawner-3*.log
[ WARN] [1744262891.652228787, 108.154000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link (parent odom) at time 108.132000 according to authority unknown_publisher
[ WARN] [1744262964.981813326, 151.031000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link (parent odom) at time 151.007000 according to authority unknown_publisher
[ WARN] [1744262975.780896362, 157.909000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link (parent odom) at time 157.909000 according to authority unknown_publisher
[ WARN] [1744263002.032932122, 173.934000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link (parent odom) at time 173.909000 according to authority unknown_publisher

Shepherd010, 2025-04-10:
Supplement: Installation based on docker's noetic image

reiniscimurs, 2025-04-10:
Hi,

The average reward and other printed info is only part of evaluation after each epoch. You probably have not encountered it yet.

There is some time synchronization issue in your implementation so the same publisher gets multiple messages with the same timestep.
You may want to change the line here:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/aba927a6a3b832a15e02183a2a9c78fce7a8732e/TD3/velodyne_env.py#L172
to: `rospy.sleep(TIME_DELTA)`
See if that helps

Shepherd010, 2025-04-10:
Thank you for your reply. May I ask if this warning is common in training? I have set it up as you said, but I still see this warning. Does this warning affect training?

Shepherd010, 2025-04-10:
Thank you for your reply. After I cancelled the startup of rviz and gazebo, no warnings were output and everything was normal. After some time, the evaluation results were also output normally.

closed 2025-04-10 by Shepherd010

reiniscimurs, 2025-04-10:
Make sure you have enough compute to run everything. If there is a graphic slowdown and the computer starts "chugging" in ros, there might be a clock desync. That could explain the timestamp issue.