reiniscimurs/DRL-robot-navigation #178: radar

LSWings, 2025-04-14:
I want to modify the radar's scanning radius. While searching through the code, I only found the setting for the radar's scanning angle. Can you offer me some suggestions

reiniscimurs, 2025-04-14:
Hi,

Which sensor do you mean? The 2d LiDAR or 3D velodyne lidar? There are no radar sensors by default in this repo.

Also, which setting do you want to modify? The maximum distance of the scans?
For 3d velodyne puck you should be able to modify it here:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/aba927a6a3b832a15e02183a2a9c78fce7a8732e/catkin_ws/src/multi_robot_scenario/xacro/p3dx/pioneer3dx.xacro#L27
You can pass in an additional argument `max_range:=130.0`

For 2d LiDAR, you can change the value here:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/aba927a6a3b832a15e02183a2a9c78fce7a8732e/catkin_ws/src/multi_robot_scenario/xacro/laser/hokuyo.xacro#L49

Do not forget to update the binning function for the velodyne callback here:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/aba927a6a3b832a15e02183a2a9c78fce7a8732e/TD3/velodyne_env.py#L137
(Change the value of 10 to the value that you have selected as max range)

LSWings, 2025-04-14:
> 2d LiDAR or 3D velodyne lidar

Thank you very much, I have already achieved the goal I needed. I want to add two small cars to achieve the objective of forming a cluster. Could you tell me what parts need to be modified？