reiniscimurs/DRL-robot-navigation #179: Issue about [gazebo-1] process has died

FanYI111, 2025-04-15:
**Describe the issue**
Hi,  when I run the command ` python train_velodyne_td3.py` in the console,  encountered the following error:

```
process[gazebo-1]: started with pid [167687]
[gazebo-1] process has died [pid 167687, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode TD3.world __name:=gazebo __log:=/home/fanyi/.ros/log/37a4e6da-19d4-11f0-acfa-cf2a52b0593d/gazebo-1.log].
log file: /home/fanyi/.ros/log/37a4e6da-19d4-11f0-acfa-cf2a52b0593d/gazebo-1*.log
```
How can I fix it, Thanks

reiniscimurs, 2025-04-15:
Sorry this is a gazebo/ros related issue and not connected to the repo itself. I wouldn't be able to help here but I suggest checking the logs to see what the issue is.

DFxiaotu, 2025-05-16:
Hi，I met the same problem, and a possible cause is that you might not run the killall command written in readme to end your last training.