reiniscimurs/DRL-robot-navigation #181: 3dvelodyne sim2real

nfhe, 2025-04-24:
**Describe the issue**
A clear and concise description of what the issue is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log output**
If applicable, full log output.

**Preliminaries**
 - Have you checked the tutorial for answers to your issue?
 - Have you checked prior issues for answer to your issue?

**Provide information about your setup:**
 - OS:
 - ROS version:
 - Repository branch:

**Additional context**
Add any other context about the problem here.

reiniscimurs, 2025-04-24:
Hi,

See the answer here: https://github.com/reiniscimurs/DRL-robot-navigation/issues/180

Generally, the test file will load model and get an action in continuous execution. In real life you will not need such functions as reset and reward calculation. So you should create your own environment class with init, velodyne_callback, odom and step methods similarly as in velodyne_env.py file. Set up your subscribers and publishers according to your robot and system and you should be good to go. As long as your input and output space is similar to what is given in the repository, you should not see any major issues.

nfhe, 2025-04-24:
Dear reiniscimurs,

I sincerely appreciate your previous response. It has been really helpful. 

I have some additional questions regarding LiDAR data. Firstly, I'm curious whether the input from 2D and 3D LiDARs has different impacts on the training process. I've noticed that many people opt for 2D LiDARs, and I'm wondering what factors contribute to this choice in terms of training.

Secondly, in the Gazebo simulation, the LiDAR range is set from -1.57 to 1.57, while in real - world LiDARs, it typically ranges from 0 to 6.28. I'm unsure if an angular transformation of the LiDAR data is necessary in such a case. 

Could you kindly offer your expertise and guidance on these matters?

Thank you very much for your time and assistance.

reiniscimurs, 2025-04-24:
There is no major difference between 2D and 3D LiDAR data in the training but the 3D lidar allows to capture more complete picture. Lets asuume a situation with an obstacle at 0.2 meter height and your 2D lidar is installed on a plane of 0.4 meters. The robot could trip over and colide with such an obstacle that would cause a different state. The 2D lidar would never the detect it but even the flattened 3D lidar would have this information. However, from the Neural network standpoint nothing changes, since it is still the same 1D input in the state.

The angle does not matter and depends entirely on how you install your lidar. The only thing that matters is that the FOV is 180 degrees in front of the robot. You can adjust the angles where this is captured accordingly.

nfhe, 2025-04-25:
Dear reiniscimurs,
﻿
Thank you very much for your detailed explanation. It has already provided me with a lot of useful information. However, I still have a few follow - up questions.
﻿
Regarding the 3D LiDAR, since the installation position of 3D LiDAR can vary, and different installation positions result in different point clouds. Does this mean that the installation position during the actual deployment needs to be the same as that during training?
﻿
As for the FOV 180 - degree angle issue you mentioned in the context of 2D and 3D LiDARs, I'm still a bit confused. What I'm specifically concerned about is that the LiDAR range in Gazebo is from - 1.57 to 1.57, while for real - world LiDARs, it's from 0 to 6.28. After clipping the real - world LiDAR data, I get data in the ranges of 0 - 90 and 270 - 359. Do I need to align the order of this real - world LiDAR data with the input order in Gazebo?
﻿
I would greatly appreciate it if you could further clarify these points for me.
﻿
Thanks again for your patience and help.

reiniscimurs, 2025-04-25:
Your lidar should be installed at a similar height and location. The policy will essentially base it's action on the distance from the obstacle. If the "dangerous" distance is different, it may cause some issues. It depends on the robot size and shape itself but in practice you do not need to be super precise.

However you align your laser does not matter, what you need to have is sequential 180 degrees from left to right in front of the robot. In this case a combination of 270-360 and then 0-90 degree range should work. Also just using 0 to 180 degrees work if you place your lidar sideways. We need a continuous sequence of points from left to right for 180 degrees, just like it is in the simulation.

nfhe, 2025-04-25:
Dear reiniscimurs,

I truly appreciate your patient and detailed responses. They have been extremely helpful to me. I will now follow your advice and conduct tests on both 3D and 2D LiDARs respectively. I'm closing this issue. Thank you once again for all your assistance.

closed 2025-04-25 by nfhe