reiniscimurs/DRL-robot-navigation #182: An abnormal issue regarding the reset of the simulation environment after using one's own model

wxy666ustb, 2025-05-08:
**Describe the issue**
     Every time the simulation environment is reset, the position of the lidar always changes. What could be the reason for this? I'm using my own car model.

     Specifically, every time the simulation environment is reset, the normal rviz should change the position of the car, but in my rviz, the car is fixed, and then the data of the lidar changes relative to the car, such as rotation. However, the release of the target point is still based on odom, which leads to the possibility that the target point may appear outside the map.
      I checked my gazebo environment and found it was normal. In the gazebo environment, it was the position of the car that changed.

**Expected behavior**
This problem has been bothering me for a long time. I hope you can help me see where the problem might arise. I'm really looking forward to your reply.

**Screenshots**

[image]

[image]

[image]

**video**

**Provide information about your setup:**
 - OS:ubuntu 20.04
 - ROS version: ROS noetic

wxy666ustb, 2025-05-08:
Let me add my velodyne_env.py

`import math
import os
import random
import subprocess
import time
from os import path

import numpy as np
import rospy
import sensor_msgs.point_cloud2 as pc2
from gazebo_msgs.msg import ModelState
from geometry_msgs.msg import Twist
from nav_msgs.msg import Odometry
from sensor_msgs.msg import PointCloud2
from sensor_msgs.msg import LaserScan  # 新增导入
from squaternion import Quaternion
from std_srvs.srv import Empty
from visualization_msgs.msg import Marker
from visualization_msgs.msg import MarkerArray

GOAL_REACHED_DIST = 0.3
COLLISION_DIST = 0.35
TIME_DELTA = 0.1

# 设出生点
# Check if the random goal position is located on an obstacle and do not accept it if it is
def check_pos(x, y):
    goal_ok = True

    if -3.8 > x > -6.2 and 6.2 > y > 3.8:
        goal_ok = False

    if -1.3 > x > -2.7 and 4.7 > y > -0.2:
        goal_ok = False

    if -0.3 > x > -4.2 and 2.7 > y > 1.3:
        goal_ok = False

    if -0.8 > x > -4.2 and -2.3 > y > -4.2:
        goal_ok = False

    if -1.3 > x > -3.7 and -0.8 > y > -2.7:
        goal_ok = False

    if 4.2 > x > 0.8 and -1.8 > y > -3.2:
        goal_ok = False

    if 4 > x > 2.5 and 0.7 > y > -3.2:
        goal_ok = False

    if 6.2 > x > 3.8 and -3.3 > y > -4.2:
        goal_ok = False

    if 4.2 > x > 1.3 and 3.7 > y > 1.5:
        goal_ok = False

    if -3.0 > x > -7.2 and 0.5 > y > -1.5:
        goal_ok = False

    if x > 4.5 or x < -4.5 or y > 4.5 or y < -4.5:
        goal_ok = False

    return goal_ok

class GazeboEnv:
    """Superclass for all Gazebo environments."""

    def __init__(self, launchfile, environment_dim):
[... 394 lines cut ...]
        marker3.pose.position.x = 5
        marker3.pose.position.y = 0.2
        marker3.pose.position.z = 0

        markerArray3.markers.append(marker3)
        self.publisher3.publish(markerArray3)

    @staticmethod
    def observe_collision(laser_data):
        # Detect a collision from laser data
        min_laser = min(laser_data)
        if min_laser < COLLISION_DIST:
            return True, True, min_laser
        return False, False, min_laser

    @staticmethod
    def get_reward(target, collision, action, min_laser, actionfactor):
        if target:
            return 100.0
        elif collision:
            return -100.0
        else:
            r3 = lambda x: 1 - x if x < 1 else 0.0
            return action[0] / 2 - abs(action[1]) * actionfactor - r3(min_laser) / 2
`

wxy666ustb, 2025-05-08:
[frames.pdf](https://github.com/user-attachments/files/20098157/frames.pdf)
Seeing your previous solution, it seems that there is a problem with the TF tree. This is my TF tree. May I ask what the problem is

wxy666ustb, 2025-05-08:
[frames.pdf](https://github.com/user-attachments/files/20098176/frames.pdf)

reiniscimurs, 2025-05-08:
Hi,

This will not be an issue in the code itself. Rather some setup in your robot files. Most likely the sources or transforms of odometry somewhere there. It has been a long time since I have looked at any of this, but check how your odom is set up in your robot and play around with the sources. For instance in this file you have odom topic, frame and source. See if you have a similar setup for your ackerman steering plugin drive: https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/catkin_ws/src/multi_robot_scenario/xacro/p3dx/pioneer3dx_plugins.xacro

wxy666ustb, 2025-05-08:
[image]
Hello, my settings should be fine, but the problem still exists in rviz

wxy666ustb, 2025-05-08:
Hello, I would like to inquire with you. Firstly, I launched Gazebo's TD3.world using a launch file, and then launched Pioneer3dx.gazebo. launch, which means loading the robot model. In Pioneer3dx.gazebo. launch, there is a sentence<arg name="robot_position" default="- x -1.0- y 5.0- z 0.01"/>that defines the initial position of the robot. I went to check the Odom topic and found that the Odom message in your program starts around - x -1.0- y 5.0- z 0.01. May I ask how this was achieved? In my understanding, the initial state of the robot determines the origin of the Odom coordinate system.
When running my program, I also used a similar command<arg name="robot_position" default="- x -1.0- y 5.0- z 0.01"/>. I found that after the program ran, the initial position of my robot was also the same as the result of your program, but the difference was that the Odom in my program completely overlapped with the initial position of the robot. My Odom message started around - x -0.0- y 0.0- z 0.01, which means that my<arg name="robot_position" default="- x -1.0- y 5.0- z 0.01"/>command moved the Odom coordinate system, but your command did not move the Odom. The coordinate system has been moved, instead of the robot's base_dank coordinate system.
The above is my confusion, and I am looking forward to receiving your reply. Thank you very much for taking the time to reply to me amidst your busy schedule!

wxy666ustb, 2025-05-08:
[image]

wxy666ustb, 2025-05-08:
Ok. Thank you for your reply. I checked the relevant Settings. Unfortunately, I didn't find any problems.
I would like to ask you. First, I started TD3.world of gazebo with the launch file, and then started pioneer3dx.gazebo.launch, that is, loaded the robot model. In pioneer3dx.gazebo.launch, there is such a sentence. Such a sentence in the following picture defines the initial position of the robot. I checked the odom topic and found that the odom message in your program was initially located around -x -1.0 -y 5.0 -z 0.01. May I ask how this was achieved? In my understanding, the initial state of the robot determines the origin of the odom coordinate system.
When running my program, I also used the same command. I found that after the program ran, the initial position of my robot was the same as the running result of your program. However, the difference was that the odom in my program completely coincided with the initial position of the robot. My odom message was around -x -0.0 -y 0.0 -z 0.01 at the beginning. That is, my command line moved the odom coordinate system, but your command line did not move the odom coordinate system. Instead, it moved the base_link coordinate system of the robot.
The above are my confusions. I'm really looking forward to receiving your reply. Thank you very much for replying to me in your busy schedule!

[image]

reiniscimurs, 2025-05-08:
So this is why I think some setup is wrong/different. If initial position defines odometry then your source of odometry is robot itself + dead reckoning (essentially the odom itself). In this repo, we use world as the source of the odometry. So all the odometry signals are returned respective to the origin of the world - that is the 0, 0 position in the middle of the map. I do not know how the files are used and the model implemented so I do not know where to look for the change here, but I suggest experimenting around and trying to find out where the odom setup is in your model and more specifically the ackerman steering plugin.

For instance, in diff drive there is a parameter 
`
    <odometrySource>1</odometrySource>`

It passes information to its plugin to use world as the source for odometry. It seems to me it is still encoder in your case. SO you should check your plugin and see of the parameters are set up correctly.

wxy666ustb, 2025-05-08:
Thank you! My problem has been solved! Thank you for solving the problem that has troubled me for a long time!

reiniscimurs, 2025-05-08:
Great to hear!

Would be great if you could share the solution so that others could find it if they encounter a similar situation. This would help out the community.

wxy666ustb, 2025-05-08:
Ok. The controller plugin I used before seemed to have some issues. Then I changed the controller plugin to the plugin in your program (a complete copy, even if mine is a four-wheel car), and changed the names of the left and right wheel coordinate systems in your plugin to the names of the two wheels at the front of my car. I found that the coordinate system problem was solved! However, there was another issue where the odom topic was not published. I then changed the plugin to another four-wheel controller of ROS and found that the problem was solved. Not only was the coordinate system fixed, but the odom topic also worked normally!
Thank you again for your help!

reiniscimurs, 2025-05-08:
Great! Thanks! That is good to know going forward

closed 2025-05-09 by wxy666ustb