reiniscimurs/DRL-robot-navigation #183: 全向小车模型的奖励函数

wxy666ustb, 2025-05-10:
Hello! After I solved the environmental problem with your help last time, I started training with my omnidirectional car. I added a dimension to the motion space and included the possibility of left and right movement. And the reward function was modified. The state space still adopts the original project. However, the training results are not as good as your original effect. You can't even do the most basic navigation. Many times, the car just spins in circles. The seed value was modified, but it didn't work either. I'm not quite sure where the problem might arise. I have the following several questions:
1. Could you help me take a look where the problem might occur? Could you help me put forward some suggestions for improvement?
2. I attempted to modify the state space and adjust the radar data to 360°. There are two solutions: One is to increase the environmental state to 40 dimensions, or perhaps it is necessary to increase the number of network layers? But I'm not sure if that would lead to insufficient performance, because I'm using a virtual machine. Secondly, considering the performance of my computer, the original 20-dimensional environment dimension is maintained, but the partition interval is changed from 9° to 18°.
I haven't modified the state space yet because I haven't given the car the possibility to move backward. So I don't think this is the reason why my training results are different from yours. But I will have this plan in the future, so I hope to get your guidance.
3. The av.Q value has been decreasing, but max Q has been increasing. Does this mean that the model training has failed?
4. What suggestions do you have for the reward function design of an omnidirectional car that can only see 180° ahead?

My training curve is as follows:

[image]

[image]

[image]

My reward function is designed as follows:

[image]

reiniscimurs, 2025-05-12:
To be honest I do not think I can provide any guidance here. The changes that you have made seem reasonable but it is also outside of the scope of my experience. Usually you would see that av.Q might decrease in the early stages of the training due to the vehicle colliding a lot and not yet figuring out what good actions are. But they should increase in time. If it does not happen then the training will not succeed. Though i cannot tell at which point you should expect it to rise again.

wxy666ustb, 2025-05-12:
Ok. Thank you. I have tried several other reward functions these days and the effect has improved a little. I will continue to try. Thank you for your reply!

closed 2025-05-12 by wxy666ustb