reiniscimurs/DRL-robot-navigation #186: figure in your paper

huangxiaobing12, 2025-05-12:
Thank you for sharing your work. I'm currently conducting research in a related field and would like to inquire about the methodology used to create this figure in your paper. Could you please provide any reference materials or technical guidance regarding its development? Your insights would be greatly appreciated.

[image]

reiniscimurs, 2025-05-12:
Hi,

Image (a) is updated from the floor plan with traced wall outlines and added representative obstacles. Image (b) is the rviz output with the recorded map and odometry. Multiple runs are recorded and the odometries are overlaid on a single map. There is really nothing to it.

huangxiaobing12, 2025-05-13:
Thank you ,I'll try it.