reiniscimurs/DRL-robot-navigation #187: The problem of the trolley swinging and stopping during the training test

t1tqq, 2025-05-19:
Dear Reinis Cimurs，
I'm currently reproducing your github code. I feel that the progress is relatively smooth, but I have also encountered some problems. I hope to receive your help.

1.I haven't modified any code in github for training. However, during the training process, it occasionally fluctuates left and right and stops moving. This phenomenon occurs very frequently during testing. I would like to ask what causes this situation. Maybe I'm not very good at deep reinforcement learning. Then, how can I solve it? Could you give a method or an idea? Thank you.

2.Another question is whether this code can be applied to four-wheel differential and Ackerman models. If possible, if I want to deploy it to the four-wheel differential or Ackerman model, is it only necessary to change the car model during training to the four-wheel differential or Ackerman model for training

The model I trained should be usable. This is the convergence of the model I trained

[image]

Hope to get your help. If you need any other information, please tell me and I will send it to you as soon as possible. Thank you

reiniscimurs, 2025-05-19:
Hi,

See similar issues here:
https://github.com/reiniscimurs/DRL-robot-navigation/issues/80

Most likely, your policy is in some local optimum and since there is no history information, it does not know this. This would appear more in evaluation due to not adding random noise to action.

Some other people report implementing an ackerman model here so it should be possible. It should be sufficient to just change the robot model and then update your velocity cap limits. The algorithm itself should be fine as it is.

reiniscimurs, 2025-05-19:
So the model has the policy that tries to optimize a returned value. In this case, you have a state and the policy then calculates which action has the highest possible value. Lets say that the action is 0 linear vel and -0.2 angular vel. The policy then gets a new state and calculates the action with the highest value for this new state. Here the outcome is 0 linear vel and +0.2 angular vel. If you execute this action you will have just returned to the previous state. This creates a loop in policy execution but the policy itself does not know that because at any given time it knows the state only from its current time step. So you end up in a cycle of going from left to right over and over again. This is more prevalent in evaluation because we never add noise to the action there so the loop is exactly the same. During training, the added random noise to the action has a slight possibility to perturb the action so much that it actually changes some state and breaks the loop.

This happens because the policy encounters some local optimum of the returned Q value for the state-action pair. The policy is greedily selecting this value for action. Here the issue is more fundamental, because the Q-value estimation is either wrong due to limited training or limited experiences of how to avoid such situations. You can simply retrain the model and hope that the newly trained critic will be able to represent the value of state-action pair better, or introduce some bootstrapping methods to give more varied experiences for the model to train on (i.e. the random_near_obstacles bootstrapping method https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/TD3/train_velodyne_td3.py#L330)

t1tqq, 2025-05-21:
Dear Reinis Cimurs,I encountered other problems when changing the Ackerman model. Could you take a look for me? Thank you.

Since I'm not very familiar with the use of gazebo, instead of modifying your model, I found an Akman model to replace the entire one, and then modified the radar parameters and namespace in it. However, problems occurred when it was running.

The following is the pioneer3dx.gazebo.launch file. The modified part is "<param name="robot description" command="$(find xacro)/xacro '$(find multi robot scenario)/xacro/ackermann/racecar.xacro'" />”
[image]

The racecar.xacro file is as follows:
<?xml version="1.0" encoding="utf-8"?>

<robot name="r1" xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:include filename="$(find multi_robot_scenario)/xacro/ackermann/macros.xacro" />
    
    <link name="base_footprint">
    </link>

    <link name="base_link">
        <visual>
            <geometry>
                <box size="0.28 0.1 0.03"/>
                
            </geometry>
            
            <origin xyz="0 0 0" rpy="0 0 0" />
            <material name="yellow">
                <color rgba="0.8 0.3 0.1 0.5" />
            </material>
        </visual>
        <collision>
            <geometry>
                <box size="0.28 0.1 0.03" />
            </geometry>
            <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0" />
        </collision>

    </link>

    <joint name="base_link2base_footprint" type="fixed">
        <parent link="base_footprint" />
        <child link="base_link"/>
        <origin xyz="0 0 0.032" rpy="0 0 0" />
    </joint>

    <link name="base_inertia">
    <inertial>
            <origin xyz="0 0 0" />
            <mass value="4" />
            <inertia ixx="0.0264" ixy="0" ixz="0" iyy="0.0294" iyz="0" izz="0.00364" />
        </inertial>
    </link>

    
    <joint name="chassis_inertia_joint" type="fixed">
    <origin xyz="0 0 0" rpy="0 0 0" />
    <parent link="base_link" />
    <child link="base_inertia" />
    </joint>

    <link name="left_steering_hinge">
        <visual>
            <geometry>
                <cylinder radius="0.01" length="0.005" />
                
            </geometry>
            <origin xyz="0 0 0" rpy="1.57 0 0" />
            <material name="black">
                <color rgba="0.0 0.0 0.0 1.0" />
            </material>
        </visual>
[... 456 lines cut ...]
/imu
/initialpose
/joint_states
/linear_velocity
/move_base_simple/goal
/r1/front_camera/camera_info
/r1/front_camera/image_raw
/r1/front_camera/image_raw/compressed
/r1/front_camera/image_raw/compressed/parameter_descriptions
/r1/front_camera/image_raw/compressed/parameter_updates
/r1/front_camera/image_raw/compressedDepth
/r1/front_camera/image_raw/compressedDepth/parameter_descriptions
/r1/front_camera/image_raw/compressedDepth/parameter_updates
/r1/front_camera/image_raw/theora
/r1/front_camera/image_raw/theora/parameter_descriptions
/r1/front_camera/image_raw/theora/parameter_updates
/r1/front_camera/parameter_descriptions
/r1/front_camera/parameter_updates
/r1/front_laser/scan
/r1/odom
/rosout
/rosout_agg
/tf
/tf_static
/velodyne_points

reiniscimurs, 2025-05-21:
seems like nothing is published in odom topic here. Can you subscribe to this topic to see if it is publishing something?

Also, maybe I missed it here, but where is your ackerman controller plugin? (like in https://github.com/reiniscimurs/DRL-robot-navigation/blob/main/catkin_ws/src/multi_robot_scenario/xacro/p3dx/pioneer3dx_plugins.xacro)

t1tqq, 2025-05-22:
I tried to modify it again, but still couldn't solve it. It seems that it's because the odom and cmd_vel topics haven't been released

First, I launched an empty environment.

[image]

Then start the pioneer3dx.gazebo.launch file.
[image]

Afterwards, the Ackerman model and liDAR were successfully loaded

[image]

I added the Ackerman controller plugin code in the racecar.gazebo file, but it seems that the plugin was not successfully loaded. Do you know where the problem lies? This is a bit challenging for me.

[image]

When I checked the topics, I found that the "odom and cmd_vel" topics were missing

topic list:
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressed/parameter_descriptions
/camera/depth/image_raw/compressed/parameter_updates
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/compressedDepth/parameter_descriptions
/camera/depth/image_raw/compressedDepth/parameter_updates
/camera/depth/image_raw/theora
/camera/depth/image_raw/theora/parameter_descriptions
/camera/depth/image_raw/theora/parameter_updates
/camera/depth/points
/camera/parameter_descriptions
/camera/parameter_updates
/camera/rgb/camera_info
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/performance_metrics
/gazebo/set_link_state
/gazebo/set_model_state
/imu
/joint_states
/r1/front_camera/camera_info
/r1/front_camera/image_raw
/r1/front_camera/image_raw/compressed
/r1/front_camera/image_raw/compressed/parameter_descriptions
/r1/front_camera/image_raw/compressed/parameter_updates
/r1/front_camera/image_raw/compressedDepth
/r1/front_camera/image_raw/compressedDepth/parameter_descriptions
/r1/front_camera/image_raw/compressedDepth/parameter_updates
/r1/front_camera/image_raw/theora
/r1/front_camera/image_raw/theora/parameter_descriptions
/r1/front_camera/image_raw/theora/parameter_updates
/r1/front_camera/parameter_descriptions
/r1/front_camera/parameter_updates
/r1/front_laser/scan
/rosout
/rosout_agg
/tf
/tf_static

Terminal information：

~/DRL-robot-navigation/catkin_ws$ roslaunch multi_robot_scenario TD3_world.launch 
... logging to /home/lw/.ros/log/2722ff02-36ff-11f0-84eb-b13eb20fd715/roslaunch-lw-26412.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[... 75 lines cut ...]
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    r1_controller_spawner (controller_manager/spawner)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[r1_controller_spawner-1]: started with pid [26658]
process[urdf_spawner-2]: started with pid [26659]
process[robot_state_publisher-3]: started with pid [26660]
process[joint_state_publisher-4]: started with pid [26661]
[INFO] [1747920421.424417, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1747920421.429121, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1747920421.432619, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1747920421.947592, 20.085000]: Spawn status: SpawnModel: Successfully spawned entity
[urdf_spawner-2] process has finished cleanly
log file: /home/lw/.ros/log/2722ff02-36ff-11f0-84eb-b13eb20fd715/urdf_spawner-2*.log
[WARN] [1747920451.567471, 49.009000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[r1_controller_spawner-1] process has finished cleanly
log file: /home/lw/.ros/log/2722ff02-36ff-11f0-84eb-b13eb20fd715/r1_controller_spawner-1*.log

reiniscimurs, 2025-05-23:
My guess would be that something is off with the controller plugin. At least I would not expect all these parameters to be posted in parameters section:
/r1/ackermann_controller/cmd_timeout: 0.5
/r1/ackermann_controller/front_steering_joints: ['left_steering_h...
/r1/ackermann_controller/publish_rate: 50
/r1/ackermann_controller/rear_wheel_joints: ['left_rear_wheel...
/r1/ackermann_controller/type: ackermann_drive_c...
/r1/ackermann_controller/wheel_base: 0.3
/r1/ackermann_controller/wheel_radius: 0.075
/r1/ackermann_controller/wheel_separation: 0.2
/r1/robot_description: <?xml version="1....

Though I do not know what is going on here so it is quite hard for me to give any guidance.

t1tqq, 2025-05-25:
Could you help me take a look at the problem that has emerged now? Thank you

I used the command "python3 train_velodyne_td3.py", which can initially train and display in rviz. However, there are some issues. Firstly, there is an error message. At the same time, the quaternion signs are inconsistent (alternating positive and negative), the rotation angle changes abruptly, and the linear velocity is abnormal. The relevant information is below. Can you figure out where the problem lies? Anyway, thank you for your help.

This error report is “[ERROR] [1748174376.134181596]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
”

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_odometry_node (racebot_control/gazebo_odometry.py)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    servo_commands (racebot_control/servo_commands.py)
    transform (racebot_control/transform.py)
    urdf_spawner (gazebo_ros/spawn_model)
  /racebot/
    controller_manager (controller_manager/spawner)

ROS_MASTER_URI=http://localhost:11311/

process[gazebo-1]: started with pid [226966]
process[urdf_spawner-2]: started with pid [226968]
process[racebot/controller_manager-3]: started with pid [226972]
process[robot_state_publisher-4]: started with pid [226973]
process[servo_commands-5]: started with pid [226974]
process[transform-6]: started with pid [226976]
process[gazebo_odometry_node-7]: started with pid [226977]
process[rviz-8]: started with pid [226981]
[INFO] [1748173065.536475, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1748173065.540035, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1748173065.608438, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1748173065.681808852]: Finished loading Gazebo ROS API Plugin.
[INFO] [1748173065.683257201]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1748173067.201562337]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1748173067.222208470]: Physics dynamic reconfigure ready.
[ERROR] [1748173067.294328190]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
[ERROR] [1748173067.294578503]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
[INFO] [1748173067.348979, 0.297000]: Calling service /gazebo/spawn_urdf_model
[ERROR] [1748173067.355162328]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
[ERROR] [1748173067.415561244, 0.351000000]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
[ERROR] [1748173067.471050773, 0.401000000]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
[ERROR] [1748173067.533264976, 0.458000000]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
Warning [parser_urdf.cc:1161] multiple inconsistent <name> exists due to fixed joint reduction overwriting previous value [Gazebo/Black] with [Gazebo/Grey].
Warning [parser_urdf.cc:1161] multiple inconsistent <name> exists due to fixed joint reduction overwriting previous value [Gazebo/Black] with [Gazebo/Grey].
[INFO] [1748173069.097334875, 0.464000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[INFO] [1748173069.099663756, 0.464000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[INFO] [1748173069.101683764, 0.464000000]: <initialOrientationAsReference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[INFO] [1748173069.101937433, 0.464000000]: <robotNamespace> set to: //
[INFO] [1748173069.101979723, 0.464000000]: <topicName> set to: imu
[INFO] [1748173069.101991385, 0.464000000]: <frameName> set to: imu
[INFO] [1748173069.102023365, 0.464000000]: <updateRateHZ> set to: 100
[INFO] [1748173069.102041179, 0.464000000]: <gaussianNoise> set to: 0
[INFO] [1748173069.102065595, 0.464000000]: <xyzOffset> set to: 0 0 0
[INFO] [1748173069.102092155, 0.464000000]: <rpyOffset> set to: 0 -0 0
[INFO] [1748173069.131175504, 0.464000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[INFO] [1748173069.132927896, 0.464000000]: Starting Laser Plugin (ns = /)
[INFO] [1748173069.134855380, 0.464000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[INFO] [1748173069.284158199, 0.464000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[INFO] [1748173069.286494894, 0.464000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[INFO] [1748173070.611306909, 0.464000000]: Velodyne laser plugin missing <min_intensity>, defaults to no clipping
[INFO] [1748173070.613313, 0.464000]: Spawn status: SpawnModel: Successfully spawned entity
[... cut ...]
            in RPY (degree) [0.000, 0.002, -53.211]
At time 40.784
- Translation: [0.921, 2.288, -0.000]
- Rotation: in Quaternion [0.000, 0.000, -0.235, 0.972]
            in RPY (radian) [0.000, 0.000, -0.474]
            in RPY (degree) [0.002, 0.005, -27.169]
At time 41.798
- Translation: [0.923, 2.287, 0.000]
- Rotation: in Quaternion [0.000, -0.000, -0.234, 0.972]
            in RPY (radian) [0.000, -0.000, -0.472]
            in RPY (degree) [0.004, -0.008, -27.019]

[image]

Do you need any other information?

reiniscimurs, 2025-05-25:
Sorry I don't think I can help with this one. This is really outside the scope of this repo and introduces a lot of changes that I would not be able to check nor would have the time to get to the bottom of.

t1tqq, 2025-05-28:
I still have a question to ask you. I tried different seeds to train the Ackerman model, but I encountered this phenomenon every time. Do you think what factors have caused this problem? I don't know where to start to solve it.

During the training period, the speed was normal at the beginning.  However, within a few episodes, the speed would consistently drop to zero. I checked the /cmd_vel topic and found it to be the case. Even after resetting the target point, the speed remained at zero. Below are a few videos. The first one shows the normal speed at the start, the second one becomes very slow, and the third video has some stuttering for unknown reasons. I only changed this training code. If it's -1, the speed will drop to zero in a shorter time.

https://github.com/user-attachments/assets/bae23aab-17e6-4e13-836c-5c8873bd7aff

https://github.com/user-attachments/assets/c20e9625-4965-4d36-84f9-95d676cb1adb

https://github.com/user-attachments/assets/34f07e69-87df-4def-85c9-9e0ba7eabaae

reiniscimurs, 2025-05-28:
How long have you been running the training for? The training to get a proper policy will take quite some time and it will look very bad in the beginning after a couple of training cycles.

The stuttering is due to adding random noise to the action during the training in order to encourage exploration of the action space:
https://github.com/reiniscimurs/DRL-robot-navigation/blob/aba927a6a3b832a15e02183a2a9c78fce7a8732e/TD3/train_velodyne_td3.py#L323-L325

A lot of this stuff is explained in the tutorial:
https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part1-installation-d62715722303

t1tqq, 2025-05-28:
I know this will take quite a long time, but this car will remain stationary with a speed of 0. This seems to be of no help to the training at all. Moreover, I discovered a very strange phenomenon. Only when the code runs to this section and the terminal outputs "Validating", the car keeps getting stuck, just like in the video. 

[image]

https://github.com/user-attachments/assets/8bff763e-fc26-40a0-8b37-fedf602c6ad1

After the evaluation is completed, The car can start training again and its speed is no longer zero. This leads to very similar evaluation results. I'm quite confused about this problem. Why is it like this?
[image]

reiniscimurs, 2025-05-28:
Yes because that is the validation stage. At this point the current state of the model is evaluated. No training is done here and no random value is added to the action as there is no need for exploration steps. The reason why the vehicle is standing is because at this time, the calculated policy is very much suboptimal. Please, read the tutorial and try to understand the code and what is happening there.

I see your training is also at only epoch 3. This is not enough to train a meaningful policy. In the original code it would take 40 epoch to start showing some good behavior. Here, I have no clue but I would train for at least 50 epochs and only then see if the policy is reasonable.